sensorium/Mozzi #18: Issues uploading to Arduino DUE

cavadagonzalo, 2015-03-19:
Hello,

I'm trying to use Mozzi on arduino DUE, the code compiles Ok, but when I try to upload it I get this error, any ideas on how to deal with this?

Arduino: 1.6.1 (Mac OS X), Board: "Arduino Due (Native USB Port)"

WARNING: library Mozzi claims to run on [avr] architecture(s) and may be incompatible with your current board which runs on [sam] architecture(s).

In file included from mozzi_basic.ino:2:0:
/Users/cavadagonzalo/Documents/Arduino/libraries/Mozzi-master/Oscil.h:24:25: fatal error: util/atomic.h: No such file or directory
 #include <util/atomic.h>
                         ^
compilation terminated.
Error compiling.

  This report would have more information with
  "Show verbose output during compilation"
  enabled in File > Preferences.

Thanks in advance for any help with this!

sensorium, 2015-03-20:
Hi, Mozzi is not ported to the Due.  The list of supported architectures is
on the main Mozzi page: https://sensorium.github.io/Mozzi/
Feel free to fork the project on GitHub and port it to your favourite
processor...
Tim

cavadagonzalo, 2015-03-27:
Thanks Tim, I think I'll be working with the UNO, your library is just too
good!!

Gonzalo Cavada Tacussis

MOB:(+56-9) 5 617 09 28
Juana de Lestonac 0107 - 4A
Providencia - Santiago
*cavadagonzalo.cl <http://cavadagonzalo.cl>*

closed 2017-11-11 by sensorium