sensorium/Mozzi #23: initializeCPUCycles and FAST error

jeffThompson, 2016-01-25:
Wonderful project, thank you so much! I get an error trying to compile my sketch, which I believe is from the Mozzi library:

    Users/JeffThompson/Documents/Arduino/libraries/Mozzi/MozziGuts.cpp: In function 'void startAudioStandard()':
    /Users/JeffThompson/Documents/Arduino/libraries/Mozzi/MozziGuts.cpp:285:9: error: 'class TimerOne' has no member named 'initializeCPUCycles'
      Timer1.initializeCPUCycles(16000000UL/PWM_RATE, FAST); // fast mode enables higher PWM rate
             ^
    /Users/JeffThompson/Documents/Arduino/libraries/Mozzi/MozziGuts.cpp:285:50: error: 'FAST' was not declared in this scope
      Timer1.initializeCPUCycles(16000000UL/PWM_RATE, FAST); // fast mode enables higher PWM rate
                                                      ^
    exit status 1
    Error compiling.

Is this an issue with the library?

sensorium, 2016-01-27:
Hi Jeff,
what version of Arduino are you using to compile?

Tim

jeffThompson, 2016-01-27:
Should have mentioned that! I'm running 1.6.6. Tried compiling for several different boards to see if that was the issue, but they all gave the same error.

tfry-git, 2018-03-24:
This is almost certainly caused by the compiler picking up the TimerOne.h header from a separately installed library rather than Mozzi's version.

Not sure on the best way out of this. TimerOne.cpp already has \#include with quotation marks instead of angle brackets, so that won't help. @jeffThompson are you still around? If so, could you try editing Mozzi\utility\TimerOne.cpp, replacing `#include "TimerOne.h"` with `#include "../utility/TimerOne.h"`?