sensorium/Mozzi #24: ESP8266 support

rolimat, 2016-08-21:
Hi!!! Thanks for this awesome library, it rocks!!! :)

Would it be possible to add support for make it work in a ESP8266 board? ESP8266 is much more powerful than Arduino ande it has WiFi and MIDI over WiFi compatibility. I think it will be a superb combination :)

What should I do for  making Mozzi compatible with ESP8266??

Thanks

tobozo, 2017-05-14:
And now there's the ESP32. Although the Arduino IDE implementation is not yet complete, it would also be a candidate.

@sensorium can we have your opinion on this ?

Thanks for the great work

sensorium, 2017-05-16:
Hi,
the thing about Mozzi is that it was written for limited 8 bit Atmel
processors.  Newer 32 bit boards don't need the 8 bit tricks and
optimisations.  They would be under-utilised by Mozzi, and the workarounds
in Mozzi would be wasted on the new processors.

That said, anything with an Arduino implementation can probably enable
Mozzi to be ported, as has been done on the eensy3.x boards, and Mozzi can
still be quite useful with more speed, memory and peripherals.

Something like this https://www.bountysource.com/ could be a way to
proceed, if Mozzi on new boards ever becomes really sought after.

Tim

tobozo, 2017-05-16:
Tim, 

ok this makes sense, I'll stick with my Atmel modules then :o)

be well

tfry-git, 2018-03-11:
ESP8266 support was added to master, just now. Please go ahead and try it. Do read the hardware specific notes near the bottom of README.md.

closed 2018-03-11 by tfry-git