sensorium/Mozzi #25: Copy pasting examples

ptbrowne, 2016-09-27:
Example codes on the documentation are not copy-pastable.

Is it by design ?

sensorium, 2016-09-29:
Not sure why that is...
But it's all in the distributed examples if you're stuck...

ptbrowne, 2016-09-29:
OK thanks, I have just seen the examples.

I found the origin of the problem, the text is in fact selectable but the selection is styled with CSS. In `screen.css` :

```
.gist .highlight *::-moz-selection,
figure.code .highlight *::-moz-selection {
    background: #fff;
    color: inherit;
    text-shadow: #fff 0 1px
}
.gist .highlight *::-webkit-selection,
figure.code .highlight *::-webkit-selection {
    background: #fff;
    color: inherit;
    text-shadow: #fff 0 1px
}
.gist .highlight *::selection,
figure.code .highlight *::selection {
    background: #fff;
    color: inherit;
    text-shadow: #fff 0 1px
}
```

Since the background of the code is also white, the selection is invisible.

sensorium, 2016-09-29:
Thanks for the feedback.  I don't have much time for web design, and not
much experience - should I change something there?   (It's possibly a
default in the authoring framework)...

closed 2018-08-11 by sensorium