sensorium/Mozzi #28: Does not compile when set to HIFI mode

Neurodynasoft, 2017-03-10:
Arduino 1.8.1
Mozzi 1.0.3rc4
Windows 10

Examples work fine in the default STANDARD_PLUS mode, but when attempting to compile any example in HIFI mode (by setting it in the mozzi_config.h file), it  produces the following error:

`In file included from C:\...\Arduino\libraries\Mozzi-1.0.3rc4\MozziGuts.cpp:13:0:

C:\...\Arduino\libraries\Mozzi-1.0.3rc4\MozziGuts.cpp: In function 'void setupTimer2()':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:104:37: error: expected unqualified-id before '(' token

 #define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )

                                     ^

C:\...\Arduino\libraries\Mozzi-1.0.3rc4\MozziGuts.cpp:443:29: note: in expansion of macro 'clockCyclesPerMicrosecond'

  period *= FrequencyTimer2::clockCyclesPerMicrosecond();

                             ^

C:\...\Arduino\libraries\Mozzi-1.0.3rc4\MozziGuts.cpp:444:2: error: 'setPeriod' is not a member of 'FrequencyTimer2'

  FrequencyTimer2::setPeriod(period);

  ^

exit status 1
Error compiling for board Arduino/Genuino Uno.`

sensorium, 2017-03-10:
Hi,
Mozzi uses its own modified version of the FrequencyTimer2 library, in the utilities folder.  It looks like you have another install of the FrequencyTimer2 library on your system which is conflicting with Mozzi's one.  It should work if you remove the other version while you use Mozzi.
The reason it was initially included with Mozzi was to avoid a list of dependencies which need to be installed for Mozzi to work.  Perhaps Mozzi's version could be renamed to stop conflicts when the original FrequencyTimer2 is already installed.  Even better would be if libraries could automatically install dependencies, but I'm not sure if this is possible yet with Arduino...

Neurodynasoft, 2017-03-10:
I removed ALL libraries except for Mozzi, closed arduino IDE and re-opened. Still same problem.

sensorium, 2017-03-10:
Thanks for your help.  I see the problem in line 444 of MozziGuts.cpp.  Please try the most recent commit.

Neurodynasoft, 2017-03-10:
Thank you. This solves one of the errors, but the others persist, so it will not compile yet. Here is the error message now:

In file included from C:\...\Arduino\libraries\Mozzi-1.0.3rc4\MozziGuts.cpp:13:0:

C:\...\Arduino\libraries\Mozzi-1.0.3rc4\MozziGuts.cpp: In function 'void setupTimer2()':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\cores\arduino/Arduino.h:104:37: error: expected unqualified-id before '(' token

 #define clockCyclesPerMicrosecond() ( F_CPU / 1000000L )

                                     ^

C:\...\Arduino\libraries\Mozzi-1.0.3rc4\MozziGuts.cpp:443:29: note: in expansion of macro 'clockCyclesPerMicrosecond'

  period *= FrequencyTimer2::clockCyclesPerMicrosecond();

                             ^

exit status 1

sensorium, 2017-03-11:
Removed incorrect FrequencyTimer2 namespace for clockCyclesPerMicrosecond().  Committed to github.  It compiles here, hope it works for you now.

Neurodynasoft, 2017-03-11:
Excellent! It does compile now. So it was just a namespace issue. I am glad it was a simple fix. 👍

closed 2018-02-15 by sensorium