sensorium/Mozzi #30: WARNING: Category 'Audio' in library Mozzi is not valid. Setting to 'Uncategorized'

pen2pen2, 2017-04-19:
Just downloaded the latest release. Unfortunately, this error shows up:

WARNING: Category 'Audio' in library Mozzi is not valid. Setting to 'Uncategorized'

Will it affect anything? If so, how can this be fixed?

sensorium, 2017-04-20:
Doesn't seem to affect anything - I don't know if it might be because
Arduino doesn't have a category for Audio?  Or something wrong in
library.properties?
Here's the library.properties file:

name=Mozzi
version=1.0.3
author=Tim Barrass and contributors as documented in source, and at
https://github.com/sensorium/Mozzi/graphs/contributors
maintainer=Tim Barrass <faveflave@gmail.com>
sentence=Sound synthesis library for Arduino
paragraph=With Mozzi, you can construct sounds using familiar synthesis
units like oscillators, delays, filters and envelopes.
category=Audio
url=https://sensorium.github.io/Mozzi/
architectures=*
dot_a_linkage=false
includes=MozziGuts.h

hectorlee, 2017-07-19:
The Arduino library does not have a "Audio" category. 

https://github.com/arduino/Arduino/wiki/Arduino-IDE-1.5:-Library-specification

> category - if present, one of these: "Display", "Communication", "Signal Input/Output", "Sensors", "Device Control", "Timing", "Data Storage", "Data Processing", "Other"

Seems like the most appropriate category should be Signal Input/Output.

sensorium, 2017-07-19:
Thanks Hector

orbitinstasis, 2017-09-10:
May i ask, if there was a fix for this, as the warning tends to be a bit of a bother.

sensorium, 2017-09-10:
Edit library.properties so it reads:
category=Signal Input/Output

as suggested by Hector in this thread.

It'll be fixed in the next release, if it's not in the most recent one.

per1234, 2018-02-04:
This was fixed by https://github.com/sensorium/Mozzi/commit/f89629862bcdd61d47f243c881832bf03c8d70f0.

closed 2018-08-11 by sensorium