sensorium/Mozzi #31: Switch pin 9 for 10

pen2pen2, 2017-04-20:
Hello!

I've tried this method for switching out pin 9 for pin 10 : https://groups.google.com/forum/#!topic/mozzi-users/p3q2QSCcuzU

When I tried an example, the audio is still coming from pin 9. Is there something that I've missed?

sensorium, 2017-04-21:
Hi,
the default config file is now AudioConfigStandardPlus.h.  I just tested
the changes with that file and it plays on pin 10.
Just to be clear for future reference:

In AudioConfigStandardPlus.h, change..

#define AUDIO_CHANNEL_1_OUTPUT_REGISTER OCR1A
to
#define AUDIO_CHANNEL_1_OUTPUT_REGISTER OCR1B

and
#define AUDIO_CHANNEL_1_PIN TIMER1_A_PIN
to
#define AUDIO_CHANNEL_1_PIN TIMER1_B_PIN

Tim

tfry-git, 2018-02-27:
As far as I understand, this (support) issue is resolved. If it is not, please re-open / re-submit.

closed 2018-02-27 by tfry-git