sensorium/Mozzi #32: Setting USE_AUDIO_INPUT to true causes NS1 Nanosynth to hang after 10s

dturner, 2017-08-24:
Steps to reproduce: 

- Change USE_AUDIO_INPUT to true in mozzi_config.h
- Upload the default sketch for the NS1 nanosynth: https://github.com/SOUNDMACHINES/NS1nanosynth_sketches/blob/master/NS1NANOSYNTH_BASE_NO_DIGIPOT/NS1NANOSYNTH_BASE_NO_DIGIPOT.ino

Expected result: 

- Sawtooth wave is played out of PIN 9 whilst the nanosynth has power

Actual result: 

- Sawtooth wave stops playing after around 10 seconds, and the device becomes unresponsive to MIDI input (no LED flashes when notes are sent)

Note: After problem has occurred to reflash the nanosynth you need to hold down the reset button, and release it only when the new sketch is shown as "uploading" in the Arduino IDE.

sensorium, 2017-08-25:
At first glance and from your description it is possible that the processor can't keep up when audio input is turned on.

I'll check if anything has gone wrong with audio input when I have a chance, but in the meantime I would recommend optimising the sketch for faster processing.

Some very slow operations are:
/ divide : try to use bit-shift approximations (for instance >>3 (/8) instead of /10, or even try float *0.1f)
map() : look at Mozzi's IntMap.h for a faster version
mtof(float) : try to use int or Mozzi's fixed-point versions, these all use look-up tables instead of calculating at run-time

To test, try carefully removing sections of the sketch (eg. set frequency with a literal number and comment out all the calculations), to see if it can continue with audio input without freezing.

dturner, 2017-08-28:
Rather than optimising the NS1 sketch I just tried the Mozzi Audio_Input example. I did this because it has a minimal amount of code in the `updateAudio` method: 

`int updateAudio(){
  int asig = getAudioInput(); // range 0-1023
  asig = asig - 512; // now range is -512 to 511
  // output range in STANDARD mode is -244 to 243, 
  // so you might need to adjust your signal to suit
  return asig;
}` 

I tested on 3 boards: NS1 nanosynth, normal Leonardo board and an UNO r3. 

On all 3 I was unable to hear any output from PIN 9.

sensorium, 2017-08-29:
Hi,

is your audio input signal in the range 0-5v (for a 5v arduino)?  At least 2.5v amplitude?

dturner, 2017-08-29:
My audio source is the nanosynth which generates a 5V square wave. The nanosynth also provides a 5V supply to the arduino.

dturner, 2017-08-29:
More info on this forum post: http://www.sound-machines.it/forums/topic/bug-when-sending-audio-data-to-arduino/

sensorium, 2017-09-05:
I've just tested the Mozzi audio input example sketch on an arduino nano 328p, and it works for me here...  

I'll try to test with the boards you mentioned when I can.

There are known issues with Mozzi on Leonardo using Arduino IDE after 1.0.5.

TIm

thomasfredericks, 2018-01-07:
> There are known issues with Mozzi on Leonardo using Arduino IDE after 1.0.5.

@sensorium is there an issue or a fix for this?