sensorium/Mozzi #34: Possible to use high-speed PWM pin 11?

ilionic, 2017-12-03:
I have a module using pin 11 for output ( phase-correct ). Current code looks like:
```
#define PWM_PIN       11
#define PWM_VALUE     OCR2A

// called in setup
void audioOn() {
  // Set up PWM to 31.25kHz, phase accurate
  TCCR2A = _BV(COM2A1) | _BV(WGM20);
  TCCR2B = _BV(CS20);
  TIMSK2 = _BV(TOIE2);
}

SIGNAL(PWM_INTERRUPT) {
  // get output value logic..
  PWM_VALUE = output;
}
```
Having hard times figuring out how to run Mozzi on it. Is it possible? Sorry if noob question or unrelated.

sensorium, 2017-12-03:
Hi Ilia,

I'm not quite sure what you're asking...
If you're using Mozzi in the usual way, you don't need to work out the interrupts and timers yourself.

If you're doing something else, you could look at audioHook() in MozziGuts.cpp.

Tim

ilionic, 2017-12-03:
Hi Tim,

Sorry if it's not clear. I have Ginkosyntese Grains and it is using PWM pin 11 ( i.e. https://github.com/attejensen/grains/blob/master/sqr_osc/sqr_osc.ino#L34 )
( using Arduino Nano )
I was looking into reconfiguring Mozzi to use pin 11 instead of suggested 9 but no luck.
```
// known_16bit_timers.h
#elif defined(__AVR_ATmega168__) || defined(__AVR_ATmega328P__)
#define TIMER1_A_PIN 9
#define TIMER1_B_PIN 10
#define TIMER1_ICP_PIN 8
#define TIMER1_CLK_PIN 5
```
Not sure whether something else need to be reconfigured in Mozzi in order to use that.

sensorium, 2017-12-03:
Mozzi uses Timer 1 to drive pins 9 and 10 (on a nano).  Timer 2 drives pins 3 and 11.  Maybe Timer2 can be used in STANDARD mode instead of Timer1 - I can't remember if it might have been that way at one stage of development...  no quick immediate fix without opening the manuals... you might be able to hack it just by searching for calls to the timer libraries in MozziGuts.cpp.

ilionic, 2017-12-05:
Thanks @sensorium, seems like requires deeper investigation of internals of Mozzi. So far I have no clues how to change to use Timer2. Closing it as I won't be able to handle it myself probably.

closed 2017-12-05 by ilionic