sensorium/Mozzi #46: Mozzi and wire.h

AlexandreMagnolc, 2018-03-20:
I want to control mozzi reading raw data, but when I include wire.h library I got thia error while compiling:
libraries\Wire\utility\twi.c.o: In function `__vector_24':

C:\Program Files (x86)\Arduino\hardware\arduino\avr\libraries\Wire\src\utility/twi.c:384: multiple definition of `__vector_24'

libraries\Mozzi-master\twi_nonblock.cpp.o:C:\Program Files (x86)\Arduino\libraries\Mozzi-master/twi_nonblock.cpp:456: first defined here

collect2.exe: error: ld returned 1 exit status

exit status 1
Error compiling for board Arduino/Genuino Uno.

Since i think there’s a conflict between the libraries, Is there any way to read raw directly using mozzi libraries or read raw data without wire.h?

tfry-git, 2018-03-20:
Not sure what you mean by "raw data", but is File->Examples->Mozzi->11.Communication->TwoWireRead_ADXL345 what you are looking for?

AlexandreMagnolc, 2018-03-20:
Wow, i think it’s exactly what I need. This is used to read I2C data.

Now is time to code to see if it works.
Thank you

closed 2018-03-20 by AlexandreMagnolc