sensorium/Mozzi #47: reading some wavetables from SD card

PantalaLabs, 2018-03-21:
Hello guys

I'm developing an oscillator that have many wavetables that dont fit into board memory.
I cna load then from SD card because this operation is made in "stand by" mode , so I can stop all oscillators and change my wavetables .
But , I cant figure out how to read a new wavetable that wasnt loaded at compile time.
Any clue ?

Thank you!!

tfry-git, 2018-03-21:
This is not currently possible with Oscil (at least not on AVR), because that assumes that the wavetable is stored in flash memory. Instead, when loading from SD card, you'll have it in RAM. If you feel adventurous, you could try modifying the CONSTTABLE_READ macro (globally, or - safer - only in Oscil.h) to simply read a value from a plain RAM address (as done for ESP8266, in hardware_defines.h).

There are currently some ideas for extensions to Oscil, and possibly an option to this effect could be implemented as well (but I'm not yet sure, whether / how that can be done).

PantalaLabs, 2018-03-22:
Thank you for the explanation.
I'll study the macro and if I found a solution , I write a topic here .

adrianfreed, 2018-03-22:
You might be able to add other methods that handle a non-const table pointer type?
Then you wouldn’t need a macro, probably just an array index.

tomcombriat, 2025-07-17:
Hi, sorry for the delayed response.

Have you tried? I had the feeling #48 would allow that but I have to say that I did not try or looked closely into it nor was implicated in it, but as it can read from RAM now…