sensorium/Mozzi #50: Compile error ESP8266 and external I2S DAC (pt8211)

zircothc, 2018-07-17:
\Mozzi\MozziGuts.cpp:378:51: error: 'i2s_lr' was not declared in this scope
  i2s_lr((uint16_t) (updateAudio() + AUDIO_BIAS), 0);

I've changed to
i2s_write_lr((uint16_t) (updateAudio() + AUDIO_BIAS), 0);

and now compliing is done but sound is a bit distorted.
MCP49xx dac feature is awesome. In fact, examples like fmsynth on atmega328 + MCP49xx sounds much better than ESP8266 + PT8211 so there must be something wrong on code, but I can't find it.

tfry-git, 2018-07-20:
Some examples to clarify my question 1: Is the distortion permanent or intermittent, pulsed, etc.? Is the chopping, ticking, etc.? Does it affect only loud sounds, or all sounds? Can it be described as high/low frequency noise added to the output, or is the output signal itself malformed?

zircothc, 2018-07-20:
Hi. I'll try to explain better and all test I've done.
1 - I think output is very loud. `i2s_write_lr(updateAudio(), 0);` did a good job but out now is so soft...I need to increase gain in my soundcard and electrical noises appear when I move the mouse. External power plug is really better than powering from PC USB but output is not clear at all. Arduino nano+MCP DAC is clear on both power situations. May be my ESP8266 quality is not very good.
Also tried AUDIO_RATE to 32768, same result.
I don't know how to scope output.
2 - In fmsynth example I add **<<4** to the return line:
`  return (int)aCarrier.phMod(modulation)<<4;`
Now sound is clear :) but...in that example, when pitch notes are high, sound is distorted. The single issue is that everything has a higher pitch compared to atmega328+mcpdac. it's not important for me.
3 - I did not tried it. I only tested PDM_VIA_I2S when I was building a drum machine [https://zircothc.wordpress.com/] and, of course I prefer I2S DAC output. 
4 - This trick and <<4 did the job :)

I'm very happy now ;)
Thank you very much.

zircothc, 2018-07-20:
Hi.
I think line is 580 (on mozzi master version). Anyway pitch is still high and...yes, this is the only issue now.
;)

zircothc, 2018-07-20:
Sorry, I forgot to mention that I've also tried `#define AUDIO_RATE 32768`
High pitchs now are cleaner than audio rate 16384, not distorted, but still same pitch higher.
Everything is tested with fmsynth example ;)

tfry-git, 2018-07-21:
Yes, I meant line 580. This is very likely still the source of the problem, but I have no idea, why it would be wrong. You could try AUDIO_RATE/2 or similar values, here, to see what happens. But I guess I may just have to finally solder in one of those PT8211s and try for myself...

PS: Pretty cool project you have there!

zircothc, 2018-07-21:
Hi. 
Combination of AUDIO RATE 32768 plus AUDIO_RATE/2 gives a good sound, the pitch is like atmega+mcpdac sound (yeah!) but now, note changes on fmsynth example are slow-slow ;)

Excuse me if I need to open a new Issue for this offtopic:
External DAC feature is amazing. I used to hack mozzi with a trick I saw in mozzi google forum years ago to use external DAC instead of PWM output, but yours is better, love quality output. But everytime I change between esp8266 and AVR I need to comment `#define EXTERNAL_DAC` to avoid compile errors (MCP49xx library uses PORT access)
I tried:
`#if IS_AVR()
#define EXTERNAL_DAC
#endif`
but it doesn't compile (error: missing binary operator before token "(")

Thank you very much.

tfry-git, 2018-07-21:
Hm, ok. I was working with the theory that all processing was simply sped up, and in fact my suggestion would have been expected to slow down both the frequency, and the note changes, alike.

But you're saying it's really just the frequency that is off. Very strange. Are you sure that Atmega+MCP-dac is playing at the correct frequency, or could it be that this is the one that needs fixing? E.g. could you cross-test with one of the "classical" output modes on AVR?

Regarding the compiler-error you get, that probably means that `IS_AVR()` has not yet been defined at that point in the sources. It should be safe to simply add `include <hardware_defines.h>` above your `#if IS_AVR()`.

sensorium, 2018-07-22:
I suspect the pitch issue could be in Oscil.h, setFreq()... (or similar in other classes)

zircothc, 2018-08-06:
Sorry, I was busy.
Well, pitch is not really an issue. May be you can close this issue.
Actually I'm triying to port 4 mozzi synths/sequencers I made (arduino nano+mpcdac) based on knob_LightLevel_FMsynth and AMsynth examples that I used to play audio textures on live show. May be ESP8266 is not the board I have to use: only one ADC and my "things" need 8 pots each. But.. a DUE board...would be nice.
Offtopic again, sorry:
I'm not an expert but I added Arduino DUE to mozzi library. It was so easy after you wrote `include <hardware_defines.h>`. Then I saw how to do that:  only need to edit 3 files and some `#if IS_DUE()` did the job. 
I used DueTimer.h
To do: May be add stereo is easy (DUE has DAC0 and DAC1) 
Wow, I'm very happy. DUE has lots of inputs vs ESP8266.

Thank you again :)

sensorium, 2018-08-09:
Would you like to add DUE compatibility for other Mozzi users too, with credit to you?  Maybe you could make a pull request for your code?  Or attach the changed files here, or in a new issue?

zircothc, 2018-08-09:
Hi. Sorry, I don't know how exactly github works, so, I better attach files I modified from current downloadable version.
At first I modified only 3 files (hardware_defines.h, MozziGuts.cpp and MozziGuts.h) but added a new AudioConfigDUE.h and also modified mozzi_analog.cpp to be able to use mozziAnalogRead() as analogRead() without issues.
Tested with some of the library examples:  Knob_LightLevel_x2_FMynth, FMsynth, PWM_Phasing, etc.
Only needed to add <<3 or <<4 on return value to get a good volume on UpdateAudio()
DueTimer library (https://github.com/ivanseidel/DueTimer) is needed.

[Mozzi_DUE_files.zip](https://github.com/sensorium/Mozzi/files/2276181/Mozzi_DUE_files.zip)

jimenaRL, 2018-11-06:
Hi !

I'm testing the ESP8266's port of mozzi _without_ external DAC on a Wemos d1 Lite. 

I had similar problems about output sound being very permanently distorded (both with `PDM_VIA_I2S` and `PDM_VIA_SERIAL`) and not at all at the right frequencies.

I managed to solve the issue by adding `<<8` to the return value of UpdateAudio(), for instance in the Sinewave.ino [example](https://github.com/sensorium/Mozzi/blob/master/examples/01.Basics/Sinewave/Sinewave.ino#L42).
It now outputs the  right frequency and also set higher amplitude to the output signal, which it also nice.

This is what you mean by 

> Did you make sure to scale the audio correctly? On the ESP you will want the output scaled to 16 bits in updateAudio().

???

Another question, it seems to me that the values for `PDM_VIA_I2S` and `PDM_VIA_SERIAL` are switch [here](https://github.com/tfry-git/Mozzi/blob/master/AudioConfigESP.h#L10-L11). Would that be possible ?

Anyway it sounds quite nice now, great work !
Thanks !!!

tfry-git, 2018-11-06:
Thanks for that insight. Yes, by "scale the audio correctly", I meant the left-shift, indeed. It's still interesting to see tonal noise in the unscaled (and thus very silent) signal. In _theory_ I would still expect a clear peak at the signal frequency. Do the audible frequencies change when changing the oscillator frequency, or are those "static" noise?

Regarding the switched values, I don't know what you mean. This just defines PDM_VIA_SERIAL, and PDM_VIA_I2S as two different arbitrary numbers, so that as a user you can set e.g.

```
#define ESP_AUDIO_OUT_MODE PDM_VIA_SERIAL
```

and the lib can have code like e.g.:

```
#if (ESP_AUDIO_OUT_MODE == PDM_VIA_SERIAL)
//...
#endif
```