sensorium/Mozzi #52: HIFI examples overflow audio range on avr

sensorium, 2018-07-27:
HIFI examples which used to output audio without major distortion now overflow.  The examples work when the audio in the sketches is right shifted >> 1.  Not sure if the old behaviour was incorrect and the examples should be adjusted, or if the newer code has introduced an error.
https://github.com/sensorium/Mozzi/blob/1f49dc113866eddb08dc3f2bec0553801a2ae6e8/MozziGuts.cpp#L499

tfry-git, 2018-07-27:
Note that this part of the code is for STM32 (see `#elif` on line 491). The AVR HIFI code is around line 780, and was not touched, recently.

I'll have to admit that I never included HIFI in my testing, so I cannot tell whether it may have been overflowing, before I started. But I do not _think_ I have touched any code for that while adding the STM and ESP ports, or merging any of the other recent PRs.

Could it have been this one from three years ago: https://github.com/sensorium/Mozzi/commit/b22a84c2290a3da9fc1c852e1d8100dc1fd27866 ? While that should not change the scale in itself, it will certainly cause values above the range to be handled badly. (`(B1000000 00000000 << 1) >> 8 == B0`, while previously `B10000000 00000000 >> 7 == B00000000 10000000`).

sensorium, 2018-07-27:
It seems that AUDIO_BITS_PER_CHANNEL is defined only in AudioConfigSTM32.h, which says

    #if not IS_STM32()
    #error This header should be included for STM32, only
   #endif

So I'm not sure why it's able to be used when compiling for avr...
(It's the first chance this year I've had to test the code so apologies if I'm misunderstanding it.)

sensorium, 2018-07-27:
Thanks Thomas.  Apologies for dumb questions, as I'm out of touch with the code after a long time adapting the Mozzi web page to work better with GitHub (which I hope will eventually make it easier for others to modify the pages).

I went back and tried the release version, then the current code again and it worked no problem, without any change... wierd.

When adding the <<1 then >>8, I assumed it to be faster then >>7 after tests years ago when >>7 produced 7 single shifts... but this may no longer be the case with more recent compilers.  Anyway, I'll put the issue down to corrupt memory or something, and hopefully do a speed comparison sometime.

closed 2018-07-27 by sensorium

tfry-git, 2018-07-27:
No problem. Glad to have you back!

My theory about the shift was probably wrong, too, as this problem would only kick in when the output is already at the topmost bit. That would seem unlikely to become relevant, as the output value would need to exceed 14bits _a lot_ in order to reach that point. Well, glad the problem went away by itself...