sensorium/Mozzi #54: Unable to install scikits.audiolab to run audio2huff.py script

Funkturm, 2018-08-25:
I have tried several routes to install _scikits.audiolab_ so that I can run audio2huff.py script and use some longer samples, but am still unable to do this.

sesselastronaut, 2021-01-29:
I can second this, tried various ways but no dress for success. Seems like the last version was maintained in 2010: [https://pypi.org/project/scikits.audiolab/#history](https://pypi.org/project/scikits.audiolab/#history)

MJLHThomassen, 2024-02-28:
+1, the installer for this requires python 2.6 which is super old and doesnt event include pip to install other dependencies like numpy. The script should probably rewritten to work with python 3.x (whatever is the current version). Also some instructions on which python version you need to run this would be helpfull.

tomcombriat, 2024-03-11:
@sesselastronaut Seems that you managed to get this working, according to the linked issue. Would you mind sharing your solution (and maybe propose a PR?)?

sesselastronaut, 2024-03-11:
Thanks for asking @tomcombriat the solution that worked for me can be found here: [https://github.com/Roger-random/mozzi_wilhelm/issues/1#issuecomment-772365759](https://github.com/Roger-random/mozzi_wilhelm/issues/1#issuecomment-772365759) hth