sensorium/Mozzi #58: ESP32 support

JacobErnst98, 2018-10-15:
Thanks for the ESP8266 Support but the ESP32 is better in every manner.

joknjokn, 2018-12-05:
Hi. Do you know how to make it work with ESP32? Or maybe a similar library with ESP32 support?
And I agree. ESP32 is dual core and more, so it would be so great to have support for this.

mangtronix, 2019-04-05:
Has anyone made progress on this? Thanks.

tfry-git, 2019-04-06:
Dear folks. Support for new architectures depends on someone working with those architectures to care enough to invest somewhere between 10 and 20 hours of work. Personally, I do own one ESP32, but I'm using it for a different project, and the ESP32 is a bit too expensive to have a couple, just in case. Yet more importantly, I do not have a personal use for Mozzi on ESP32, currently.

You can either: a) Try to do this yourself. It is not really all that hard. Since Mozzi has been ported to several different architectures, already, you should be able to figure out the basic steps, involved. One of the hardest parts is testing, actually. b) Offer a bounty for this, hoping somebody else will do the work for you.

mangtronix, 2019-04-12:
I have basic sound output working on ESP32 using the ESP8266Audio library (https://github.com/earlephilhower/ESP8266Audio). That library is licensed GPL, so I believe it's not possible to integrate with Mozzi because the Mozzi Creative Commons license is not compatible. Is there any possibility of making Mozzi dual license with a GPL compatible license?

zircothc, 2019-05-10:
Hello. I ported Mozzi to Arduino Due with a very good result. It was really easy.
Now I've ported Mozzi to ESP32 I2S DAC (PT8211) but audio output is a bit dirty. 
May be timer issue. I'm very busy next days and cannot test it . If somebody want it, library patched files are in zip attached included DUE patch (DueTimer library needed to run).
[Mozzi_ESP32_and_DUE_files.zip](https://github.com/sensorium/Mozzi/files/3168326/Mozzi_ESP32_and_DUE_files.zip)

tobozo, 2019-06-20:
looks like another angle https://github.com/bitluni/ULPSoundESP32

mangtronix, 2019-06-20:
Maybe he needs that for his application (I didn't watch the video) but I
have a hard time believing that sending audio to the DAC via DMA (the usual
way on ESP32) isn't the right answer. There's not that much needed for an
audio driver... basically you need to set up the I2S and then fill a ring
buffer with the audio samples. The samples in the ring buffer can be sent
to the DAC in the background, clocked by the DAC itself.

tobozo, 2019-06-20:
I've only had acceptable sound using an external DAC or the ULP, the Tetris demo is very convincing

wyan, 2020-05-05:
Did anyone manage to do some progress on ESP32 support lately?

mangtronix, 2020-05-05:
I got sound output working using this library (https://github.com/earlephilhower/ESP8266Audio), but unfortunately the licenses are not compatible so I can't release the code. In the end it wasn't toooo hard... basically you just send the samples to the audio library and you can cut out a lot of the more esoteric Mozzi code.

wyan, 2020-05-05:
Thanks! I'm not as familiar with the internals of Mozzi as I'd like, so I'm not sure I follow you. Perhaps posting a code patch without merging it could be a way to respect both licences? That way it's the end user who's applying the patch and that's ok as long as they are not distributing the code or a product derived from it?

zircothc, 2020-05-06:
> Did anyone manage to do some progress on ESP32 support lately?
Here yo can find patched files to support ESP32 with a very good sound.
https://groups.google.com/forum/#!topic/mozzi-users/qw2nIkcVaZE

tfry-git, 2021-01-08:
ESP32 is now supported in the main lib with several output modes.

closed 2021-01-08 by tfry-git