sensorium/Mozzi #59: Missing License.txt

majikthys, 2018-11-08:
What license is this code distributed under?

renzenicolai, 2020-12-20:
Would it be an option for you to release this software under a more software oriented license like GPLv2, BSD or MIT? Especially the "non commercial" part of your current license makes it impossible to use your software in projects like event badges (https://hackaday.com/?s=badge), while the "share alike" part actually forbids anyone from making changes to your code, going directly against the fundamentals of true open source software.

czho, 2020-12-20:
I feel the same way. Also if anyone really wanted to just flat out take all of mozzi and not credit anyone then they could. what are you guys going to do sue them? Law suits cost alot of money and realistically no one here has that kinda money to do anything to stop it. I feel that licences just stops honest people who wana make projects like that from being able to do so.

sensorium, 2020-12-21:
If you have a commercial project using Mozzi you are welcome to ask me about a commercial license for your project.  Usually when people ask about selling a product using Mozzi I suggest something like the first hundred units are royalty-free and then maybe something like 1% of the sale price as a royalty.  I'm also open to other proposals.  The idea is to encourage creative small scale production and to legally protect my work from large scale commercial exploitation.  
The Creative Commons Non-commercial Share-alike license (as I understand it) means that unless you and I agree on something else, you are free to use my work in your project without asking me as long as you share your work with others for free, as I have shared with you, and that you also ask the same of them.

closed 2022-03-31 by majikthys