sensorium/Mozzi #62: MOZZI Library with Teensy LC

crystalcave, 2018-11-10:
Hi to all,
I have read in the documentation that the test on the Teensy LC has been carried out with positive result, but it does not result anywhere how to configure it.
If I try to follow the documentation regarding the changes needed to use a Teensy, I have a lot of compiling problems.

The wording on the https://sensorium.github.io/Mozzi/ page also speaks of Teensy LC. Is this an error or can I not find the specific documentation?

sensorium, 2018-11-11:
Hi,
does it work with the development code from github (rather than the release)?  The reference to LC has been added by another developer, so I'm not sure without a more in-depth look at the changes...
Tim

bptremblay, 2022-11-10:
I would like to use Mozzi with the LC but the include for teensy does not seem to be aware of LC, which is the only M0 teensy. Compilation errors suggest that it thinks it's a Teensy 2 family board:

In file included from /Users/btremblay/Documents/Arduino/libraries/Mozzi/MozziGuts.h:193:0,
                 from /Users/btremblay/Sinewave/Sinewave.ino:19:
/Users/btremblay/Documents/Arduino/libraries/Mozzi/AudioConfigTeensy3_12bit.h:4:2: warning: #warning If you get a compilation error you should probably update Teensyduino to its latest version [-Wcpp]
 #warning If you get a compilation error you should probably update Teensyduino to its latest version
  ^
In file included from /Users/btremblay/Documents/Arduino/libraries/Mozzi/MozziGuts.h:193:0,
                 from /Users/btremblay/Documents/Arduino/libraries/Mozzi/MozziGuts.cpp:15:
/Users/btremblay/Documents/Arduino/libraries/Mozzi/AudioConfigTeensy3_12bit.h:4:2: warning: #warning If you get a compilation error you should probably update Teensyduino to its latest version [-Wcpp]
 #warning If you get a compilation error you should probably update Teensyduino to its latest version
  ^
/Users/btremblay/Documents/Arduino/libraries/Mozzi/mozzi_rand.cpp:148:2: warning: #warning Automatic random seeding not implemented on this platform [-Wcpp]
 #warning Automatic random seeding not implemented on this platform
  ^
/Users/btremblay/Library/Arduino15/packages/teensy/hardware/avr/1.57.1/libraries/ADC/AnalogBufferDMA.cpp: In member function 'void AnalogBufferDMA::init(ADC*, int8_t)':
/Users/btremblay/Library/Arduino15/packages/teensy/hardware/avr/1.57.1/libraries/ADC/AnalogBufferDMA.cpp:175:26: error: call to 'ADC::enableDMA' declared with attribute error: Use adc->adcX->enableDMA instead
   adc->enableDMA(adc_num);
                          ^

exit status 1

Compilation error: exit status 1

tomcombriat, 2022-11-10:
Just tested. I worked with the previous version of Teensyduino, but, after updating, I get the same error.
Actually, the error is in the ADC library of Teensyduino (or more precisely, the ADC library embedded in Teensyduino), because the following sketch:

`#include <ADC.h>
void setup() {}
void loop() {}`

Does not compile…

There is actually an (old!) PR pending on this issue: https://github.com/pedvide/ADC/pull/81
A quick fix, for you, would be to apply the patch proposed in the PR to your library. I put a comment there, hopefully this will be fixed in the teensyduino library.

Will close this old issue soon as the source of the error is not Mozzi…

Hope it helps,

bptremblay, 2022-11-11:
@tomcombriat Thanks for taking time for this!
Do I need to ping pjrc to get the adc library working?

bptremblay, 2022-11-11:
(meanwhile I will use some other boards)

tomcombriat, 2022-11-11:
@bptremblay If you have the time I would be grateful. I commented on the ADC library but I am not sure a lot happens there. Probably pjrc has their own fork anyway.

Closing this as the issue is not related to Mozzi.

closed 2022-11-11 by tomcombriat

tomcombriat, 2022-11-11:
Note:
The ADC lib has been updated. Hopefully, this will get through pjrc next version. https://github.com/pedvide/ADC/pull/81