sensorium/Mozzi #67: Strange frequency shift on STM32

Tinwelint78, 2019-04-03:
I have identified a strange problem running mozzi on STM32. Playing a Sample or an Oscil is slightly faster than it should be (exactly 1.5% faster). For example, if I make a sketch that plays a sinewave at 440 Hz, on the AVR I obtain a correct 440 Hz sinewave, the same sketch running on a STM32 produces a 447Hz sound.
I verified this is independent from audio mode (HIFI or STANDARD_PLUS), and independent from the actual wavetable used.

tfry-git, 2019-04-03:
Very interesting. Possibly our audio output timer is slightly off.

Some questions:
1) Am I right in guessing that the sample also plays 1.5% *faster*?
2) Have you tried different ```AUDIO_RATE```s?
3) What STM32 is this, exactly? Any chance F_CPU is not set quite correctly? Are you seeing the same thing on other boards of the same or different type (if available)?
4) Perhaps you might want to take a look at https://github.com/sensorium/Mozzi/blob/53889db0d6152083412bf74a85bc7aa7f32e24d5/MozziGuts.cpp#L523 . Possibly we run into a case of bad rounding, there.

Tinwelint78, 2019-04-03:
1. You are right there.
2. Now I have tried 16384 and the frequency is near exact... See point 4!
3. It is STM32F103C at 72 MHz, a classic "Blue Pill" board. I don't have another type to try right now, but I don't think the problem is there...
4. Well spotted! the problem is indeed a bad rounding:
1000000/16384 = 61.03515625 => rounded to 61, error of less than 0.05%
1000000/32768 = 30.517578125 => rounded to 30, error of 1.7%!
I think it will be better to set that timer prescaler and overflow manually instead of using setPeriod function. I will look into that and propose a PR in the next few days.

tfry-git, 2019-04-04:
Thanks! Looking forward to that!

closed 2019-04-05 by tfry-git

tomcombriat, 2021-10-03:
Hi,

I think the Teensy are also subject to this. Will try to append this correction to #137 .

PR #69 (2019-04-05): Manually calculate prescaler and overflow for STM32 update timer
Fixes #67