sensorium/Mozzi #68: Gemma M0 support

R4ph3rd, 2019-04-03:
Hi,
it seems that Mozzi has trouble to support Gemma M0 boards. Compilation aborts because the file <util/atomic.h> is missing. As it has been tested and verified as working, I tried with others release, without any improvement.
Nice tool btw !

tfry-git, 2019-04-04:
Hi could you give the exact error message (with file and line of the problem, importantly)?

Are you running the git version, or a release?

R4ph3rd, 2019-04-04:
Of course. Actually, I runned it with the lastest release 1.0.2, but also tried  with this one avaible on Mozzi page (2013 version), and the latest pre-release.

`In file included from /home/r4ph3rd/Arduino/libraries/Mozzi-1.0.2/examples/01.Basics/Control_Gain/Control_Gain.ino:19:0:
/home/r4ph3rd/Arduino/libraries/Mozzi-1.0.2/Oscil.h:24:25: fatal error: util/atomic.h: No such file or directory
 #include <util/atomic.h>
compilation terminated.
exit status 1`

tfry-git, 2019-04-04:
Thanks. The note about Gemma M0 support seems to have been added along with documentation for the Teensy port (https://github.com/sensorium/Mozzi/commit/c5aef3c4c14d793f8c754fc8006a699f090038ca). @adrianfreed any chance the Gemma M0 is supported by the Teensy core (which has atomic.h) in Mozzi?

(atomic.h is *probably* not even needed, esp. since the control loop is no longer running in an ISR, these days, but I sure am curious, where the note came from).

b2renger, 2019-04-04:
I have pretty much the same error playing with the circuit playground express which is also marked as supported. Also tried with the latest stable release and several pre-release with latest version of arduino IDE. (and also thanks a lot for the awesome lib !)

b2renger, 2019-04-08:
@tfry-git  @R4ph3rd Did a bit of looking around, and found out that the support for teensy / gemma / playground express is quite new and is not in the latest release. So taking the latest code base from master everything compiles and works fine.

Will do some test with the gemma tomorrow when I get my hands on it but I think you can close this one as it will be fixed with the next release.

tfry-git, 2019-04-09:
Thanks for the update! Closing.

closed 2019-04-09 by tfry-git