sensorium/Mozzi #70: Configure AUDIO_MODE in sketch

sensorium, 2019-04-06:
I've found a way to configure AUDIO_MODE in sketches instead of editing mozzi_config.h.  AUDIO_MODE will be STANDARD_PLUS by default.  Including a file called hifi.h in the first line of the sketch will set it to HIFI.  

Old HIFI sketches would need to be updated, but it's a simple fix which a compiler warning could also advise.

Would it be better to require the config file also for STANDARD_PLUS, so AUDIO_MODE is always explicit in the sketch?  It would break more sketches but is more consistent.

Any suggestions are welcome.  

If there are no objections I'll go ahead as I think it's a good improvement in usability.

tfry-git, 2019-04-06:
Good idea, indeed!

Breaking all STANDARD_PLUS sketches would seem a rather heavy collateral to me. But perhaps you can make it so that 
- including one of the "mode" header, explicitly, becomes documented best practice
- if no audio header has been included, Mozzi will default to "standard_plus.h" with a warning

Does this work for AUDIO_RATE, too? Can you set the two independently using a define near the top of your sketch?

sensorium, 2019-04-19:
After testing, it didn't work after all - the #defines included at sketch level don't work in the source files, even if included above the sources in the sketch.  It looked like it was working, until I listened to audio output.
I reckon I went through this years ago, and just forgot.

closed 2019-04-19 by sensorium