sensorium/Mozzi #72: Dual-license with GPL?

mangtronix, 2019-04-12:
Is it possible to make Mozziavailable under GPL license? That would make it possible to combine with GPL licensed code. The CC-BY-NC-SA is not compatible with GPL.

Creative Commons itself "[recommends and uses free and open source software licenses for software.](https://creativecommons.org/about/program-areas/software/)"

I want to publish ESP32 support for Mozzi using the ESP8266Audio library, which is GPL licensed. Mozzi could add the GPL as an option in addition to the existing license.

sensorium, 2019-04-19:
My understanding is that if your application is non-commercial you can share Mozzi under CC-BY-NC-SA.  What is the hurdle to combining with GPL code?  

If you have a commercial application, we can negotiate a license.

tfry-git, 2019-04-21:
I believe the problem is the following:
- If you create a work using ESP8266Audio, then you *must* license that under GPL.
- If you are create a work using Mozzi, then you *must* (in be absence of another agreement) license that under CC-BY-NC-SA.
- Although the two are similar in spirit, they are not compatible, thus you have no option to create a work using *both* ESP8266Audio *and* Mozzi.

One way out would be, if Mozzi would *optionally* allow you to distribute your derived work under GPL. I'd be sympathetic to that idea, but clearly it's your call, alone, to make.

tfry-git, 2019-04-21:
I will add that the GPL does not preclude commercial use. All it says is that you must offer your source code under the GPL.

DCooper-nz, 2019-04-29:
Would be great if this could be worked out as esp32 support would be fantastic.

wyan, 2021-01-04:
What's the reason for the ESP8266 port not requiring ESP8266Audio but the ESP32 requiring it?

tfry-git, 2021-01-04:
That comment was before support for the ESP32 was added to Mozzi (very recently). Neither port requires ESP8266Audio. Only the code the OP seemed to have in mind, did. The problem about a license clash if using both libs in the same project still stands.

sensorium, 2021-05-14:
Hi, it's taking a while but I've been looking at licenses and I don't really understand how a dual license works, or how to present it with a project.  Is there an example somewhere?  Thanks...

wyan, 2021-05-17:
The cases I know are products which are dual licenced GPL and a commercial licence. In those cases the customer can decide to use the GPL version for free, or, if they need to use the product in a commercial product that they don't want falling under the GPL, pay a licencing fee to obtain a non-GPL version. Some GPL products are kind of dual licenced as well, as they are licenced under "Version 2 of the GPL or any later version", at the choice of the user.

tfry-git, 2021-05-17:
I believe the standard procedure is to provide separate license files (i.e. LICENSE.GPL and LICENSE.CC-BY-NC-SA) for the respective "standalone licenses". Then you'd create a third file LICENSE which contains something like

> This work is dual-licensed under the terms of the CC-BY-NC-SA or alternatively under the terms of the GPL 2.0 (or any later version). Copies of these are distributed with the Mozzi sources as LICENSE.GPL and LICENSE.CC-BY-NC-SA, or can be found at _upstream-license-url_. You may use Mozzi according to either of these licenses as is most appropriatefor your project. (For - paid - commercial licensing options contact @sensorium.)

That short text (refined at bit) what also be included in the places where the license is stated right now, importantly file headers and README.

See also https://softwareengineering.stackexchange.com/questions/181040/how-to-document-a-dual-open-source-license , which has further variants and links to examples.

eclab, 2023-11-11:
I just posted here as well:  https://github.com/sensorium/Mozzi/issues/208

In short: CC-BY-NC-SA is not only a proper open source license, it is fully incompatible with GPL and almost certainly incompatible with LGPL.  The Arduino top-level is LGPL and certain other elements are GPL.

This means that in all probability it is a license violation to use Mozzi with any Arduino libraries.  That ain't good.

A dual license won't cut it: you simply cannot use CC-BY-NC-SA with the Arduino.  Please relicense Mozzi with a proper open source license, I personally suggest Apache 2.0, but GPLv3 would work fine.

tfry-git, 2023-11-11:
To clarify, I do not see a reason why it should be illegal to use an LGPL licensed lib from within Mozzi or to use both Mozzi and an LGPL lib in a project. The latter would then have to inherit CC-BY-NC-SA, though.

Things that do not currently work (and I agree, that is bad):
- Mixing Mozzi in any which way with GPL-only libs / projects
- Using Mozzi in a project or lib that is not CC-BY-NC-SA licensed -> makes it hard to build anything on top, even free stuff

I will add that "non-commercial" may in fact not be the boundary that really counts: Suppose you want to sell a fully open (hardware+software) DIY-kit for a customizable synth based on Mozzi. That's not currently allowed. My feeling is that "open" is the important thing, not "non-commercial".

Finally, yes, dual-licensing (or triple-, quadruple-) does help. It allows the user to pick the suitable license among the offered options. Should any of the offered licenses not be usable in the resulting combined work, this is not a problem at all: Dual-licensing means to allow a logical "OR", not force a logical "AND".

In fact, should the decision be to move to GPL, dual-licensing will be *mandatory*, in order not to break existing (law-abiding) projects which are forced to use CC-BY-NC-SA, and may or may not be capable of switching. (Should the decision be to move to LPGL or APACHE/BSD, whatever, it would not be strictly necessary to keep CC-BY-NC-SA among the set of licence options).

eclab, 2023-11-12:
The main problem with CC-BY-NC-SA + LGPL is not that LGPL is viral, but that *CC-BY-NC-SA* is viral, and also that CC-BY-NC-SA is not an open source software license and so does not properly use open source terminology.

CC-BY-NC-SA says that if you "If you remix, transform, or build upon the material, you must distribute your contributions under the same license as the original."  It doesn't explain what "build upon" means in a coding context, because CC-BY-NC-SA is not meant for code: so we are required to interpret it broadly.  A reasonable interpretation is that if you use an LGPL library in your CC-BY-NC-SA, you have "built upon" (certainly extended) the CC-BY-NC-SA project.  But that requires changing the LGPL code's license to CC-BY-NC-SA, which is not permitted.

GPL + LGPL work because LGPL explicitly can be changed to GPL.

Apache + LGPL or BSD LGPL or MIT + LGPL or CC0 + LGPL work because they don't change LGPL.

tfry-git, 2023-11-12:
I believe, we're drifting away from the (valid) topic, but:
> But that requires changing the LGPL code's license to CC-BY-NC-SA, which is not permitted.

No, it does not. It requires using CC-BY-NC-SA as the license of the **derived** work, which is allowed by the LGPL (assuming no modifications in the LGPL-code, itself).

eclab, 2023-11-12:
I appreciate the argument but respectfully disagree with it.

First to nitpick, CC-BY-NC-SA does not use the term "derived work", but uses the term "Adapted Material", notionally "derived from or based upon" the "Licensed Material" (unfortunate because "derived work" is well defined by the US Copyright Office).

In v3.0 of the licenses, there was an exception carved out for collections. That exception does not exist any more in 4.0, and instead it just says "translated, altered, arranged, transformed, or otherwise modified in a manner requiring permission under the Copyright and Similar Rights held by the Licensor".  This verbiage is clearly intended for things other than code.  Indeed "Performances" and "Sound Recordings" are also considered adapted material.

In the process of compilation, the licensed code is translated and transformed, and in the process of linking, it is also arranged with the library code.  Whether this falls under the Adapted Material clause is not clear, though the GPL has historically argued that linking is transformation.

The fundamental problem is that CC-BY-NC-SA was not meant for code, and so it has legally murky and unspecific language which can be interpreted in different of ways when it comes to linking etc.  The common safety guidance for reading licenses is to assume the broadest scope.  I argue that CC-BY-NC-SA should be considered incompatible with LPGL because of its viral nature and the murkiness of its text.

Mozzi simply needs a better license.

tfry-git, 2023-11-12:
> Mozzi simply needs a better license.

I don't disagree on that.

Regarding compatibility with (LPGL) Arduino core/license, see their FAQ: https://support.arduino.cc/hc/en-us/articles/4415094490770-Licensing-for-products-based-on-Arduino : "If you developed a sketch based on Arduino and you want to publish it as an open-source project in source code form, no restriction applies."

(As, however, Mozzi does impose restrictions, CC-BY-NC-SA is currently the only possible license available, in this case, as things stand. As for your - far reaching - theory that this would somehow need to extend to the upstream LGPL source code, too, do differentiate carefully, between what CC-BY-NC-SA calls "Adapted Material" and the more narrow scope of where "Adapter's License" is imposed.)

As for distribution in compiled form, this is in fact more complicated (but not impossible, IMO), and some (but not all) of this complexity would be eased by a different choice of license.

eclab, 2023-11-12:
It's worth mentioning that the FAQ statement does not apply to Mozzi: CC-BY-NC-SA is not an open source license.

tfry-git, 2024-06-12:
Mozzi 2.0 moved to LGPL. Closing.

closed 2024-06-12 by tfry-git