sensorium/Mozzi #73: Problem with AudioDelayFeedback Example Sketch

Capobuf, 2019-06-21:
Hi all,

I've tried to use the AudioDelayFeedback example on GitHub for implement my code with a delay.
So i want to learn the basic form of delay.

Using the AudioDelayFeedback example i would like to create a sketch that only reproduce the signal delayed after release the button.
When i press the button, i want only signal in my ear.

If i try to use the delay only option in the output part of the code, i recive an error.

The error is:

```
testdelay:57:47: error: call of overloaded 'next(int8_t, byte&)' is ambiguous

   return aDel.next(aTriangle.next(), del_samps); // instead of the previous 2 lines for only the delayed signal

                                               ^

In file included from C:\Users\User\Documents\Arduino\testdelay\testdelay.ino:18:0:

C:\Users\User\Documents\Arduino\libraries\Mozzi-master/AudioDelayFeedback.h:92:10: note: candidate: int16_t AudioDelayFeedback<NUM_BUFFER_SAMPLES, INTERP_TYPE>::next(int8_t, uint16_t) [with unsigned int NUM_BUFFER_SAMPLES = 128u; signed char INTERP_TYPE = 0; int16_t = int; int8_t = signed char; uint16_t = unsigned int]

  int16_t next(int8_t input, uint16_t delaytime_cells)

          ^

C:\Users\User\Documents\Arduino\libraries\Mozzi-master/AudioDelayFeedback.h:117:10: note: candidate: int16_t AudioDelayFeedback<NUM_BUFFER_SAMPLES, INTERP_TYPE>::next(int8_t, Q16n16) [with unsigned int NUM_BUFFER_SAMPLES = 128u; signed char INTERP_TYPE = 0; int16_t = int; int8_t = signed char; Q16n16 = long unsigned int]

  int16_t next(int8_t input, Q16n16 delaytime_cells)

          ^

exit status 1
call of overloaded 'next(int8_t, byte&)' is ambiguous
```

My code is:
```
/*  Example of flanging,
    using Mozzi sonification library.
    Demonstrates AudioDelayFeedback.
    Circuit: Audio output on digital pin 9 on a Uno or similar, or
    DAC/A14 on Teensy 3.1, or
    check the README or http://sensorium.github.com/Mozzi/
		Mozzi documentation/API
		https://sensorium.github.io/Mozzi/doc/html/index.html
		Mozzi help/discussion/announcements:
    https://groups.google.com/forum/#!forum/mozzi-users
    Tim Barrass 2012-13, CC by-nc-sa.
*/

#include <MozziGuts.h>
#include <Oscil.h>
#include <tables/triangle_analogue512_int8.h> // wavetable
#include <tables/triangle512_int8.h> // wavetable
#include <AudioDelayFeedback.h>
#include <mozzi_midi.h> // for mtof

#define CONTROL_RATE 128 // Hz, powers of 2 are most reliable

Oscil<TRIANGLE_ANALOGUE512_NUM_CELLS, AUDIO_RATE> aTriangle(TRIANGLE_ANALOGUE512_DATA); // audio oscillator
Oscil<TRIANGLE512_NUM_CELLS, CONTROL_RATE> kDelSamps(TRIANGLE512_DATA); // for modulating delay time, measured in audio samples

AudioDelayFeedback <128> aDel;

// the delay time, measured in samples, updated in updateControl, and used in updateAudio
byte del_samps;

void setup(){
  startMozzi(CONTROL_RATE);
  aTriangle.setFreq(mtof(48.f));
  kDelSamps.setFreq(.163f); // set the delay time modulation frequency (ie. the sweep frequency)
  aDel.setFeedbackLevel(-111); // can be -128 to 127
}

Q16n16 deltime;

void updateControl(){
  // delay time range from 0 to 127 samples, @ 16384 samps per sec = 0 to 7 milliseconds
  del_samps = 64+kDelSamps.next();

  // delay time range from 1 to 33 samples, @ 16384 samps per sec = 0 to 2 milliseconds
  //del_samps = 17+kDelSamps.next()/8;

  //deltime = Q8n0_to_Q16n16(17) + ((long)kDelSamps.next()<<12);

}

int updateAudio(){
  char asig = aTriangle.next(); // get this so it can be used twice without calling next() again
  //return asig/8 + aDel.next(asig, del_samps); // mix some straight signal with the delayed signal
  return aDel.next(aTriangle.next(), del_samps); // instead of the previous 2 lines for only the delayed signal
  //return asig/8 + aDel.next(asig, deltime); // mix some straight signal with the delayed signal
}

void loop(){
  audioHook();
}
```

How i can fix it?

sensorium, 2019-06-22:
Try this (cast to int8_t):

    return aDel.next((int8_t) aTriangle.next(), del_samps);

Capobuf, 2019-07-04:
I tried your suggestion, but it's the same stuff :(

My error is:

```
 C:\Users\User\Documents\Arduino\testdelay\testdelay.ino: In function 'int updateAudio()':

test_delayfeedback:57:56: error: call of overloaded 'next(int8_t, byte&)' is ambiguous

   return aDel.next((int8_t) aTriangle.next(), del_samps); // instead of the previous 2 lines for only the delayed signal

                                                        ^

In file included from  C:\Users\User\Documents\Arduino\testdelay\testdelay.ino:18:0:

C:\Users\Fabio Capo\Documents\Arduino\libraries\Mozzi-master/AudioDelayFeedback.h:92:10: note: candidate: int16_t AudioDelayFeedback<NUM_BUFFER_SAMPLES, INTERP_TYPE>::next(int8_t, uint16_t) [with unsigned int NUM_BUFFER_SAMPLES = 128u; signed char INTERP_TYPE = 0; int16_t = int; int8_t = signed char; uint16_t = unsigned int]

  int16_t next(int8_t input, uint16_t delaytime_cells)

          ^

C:\Users\Fabio Capo\Documents\Arduino\libraries\Mozzi-master/AudioDelayFeedback.h:117:10: note: candidate: int16_t AudioDelayFeedback<NUM_BUFFER_SAMPLES, INTERP_TYPE>::next(int8_t, Q16n16) [with unsigned int NUM_BUFFER_SAMPLES = 128u; signed char INTERP_TYPE = 0; int16_t = int; int8_t = signed char; Q16n16 = long unsigned int]

  int16_t next(int8_t input, Q16n16 delaytime_cells)

          ^

exit status 1
call of overloaded 'next(int8_t, byte&)' is ambiguous
```

For the rest my code it's the same, except the output part:

```
int updateAudio(){
  char asig = aTriangle.next(); // get this so it can be used twice without calling next() again
  //return asig/8 + aDel.next(asig, del_samps); // mix some straight signal with the delayed signal
  return aDel.next((int8_t) aTriangle.next(), del_samps); // instead of the previous 2 lines for only the delayed signal
  //return asig/8 + aDel.next(asig, deltime); // mix some straight signal with the delayed signal
}
```

sensorium, 2019-07-04:
Ok!  I read the error messages again.  

This compiles for me:

    int updateAudio(){
      return aDel.next(aTriangle.next(), (uint16_t)del_samps);
    }

This was the clue:
     int16_t next(int8_t input, uint16_t delaytime_cells)

sensorium, 2019-07-22:
fixed in repo

closed 2019-07-22 by sensorium