sensorium/Mozzi #75: Adafruit Feather M0 degraded audio issue

gregsadetsky, 2019-07-16:
Hello,

I've successfully ran most Mozzi examples on an Uno. I am however having problem running the same code on an Adafruit Feather M0 (which uses an ATSAMD21G18 ARM Cortex M0 processor).

The audio does come out but it seems distorted. It was pretty clear when running the Sinewave example, but it became really obvious when running the Samples Tables Arrays examples (I'm running all examples as is).

Please see below (all recordings done on the same computer using the same USB interface):
- Uno recordings: [Sinewave](https://greg.technology/_files/uno-recordings/uno-sinewave.m4a), [Samples Tables Arrays](https://greg.technology/_files/uno-recordings/uno-samples-tables-arrays.m4a)
- Feather M0 recordings: [Sinewave](https://greg.technology/_files/feather-recordings/feather-sinewave.m4a), [Samples Tables Arrays](https://greg.technology/_files/feather-recordings/feather-samples-tables-arrays.m4a)

Is there an obvious fix, or at least a direction that I could try looking into? Is this related to the onboard DAC (one of its settings)..?

Thank you very much!

tfry-git, 2019-08-01:
Did you scale the audio to 12 bits on teensy? Otherwise it will simply come out too silent, and with a worse signal-to-noise ratio.

tfry-git, 2019-08-01:
Oh, sorry, did not read well. You are not using a teensy.

I don't think the SAMD21-support has received a lot of testing so far, and I'm not sure, what board it was originally developped, for. I still suspect an audio scaling issue, though 12 bits may *not* be the correct scaling to use. (The adafruit page mentions a 10bit DAC, but the code seems to assume 12 bits).

Experiment with scaling your audio output to 8, 10, and 12 bits, testing what sounds best. Also, you may have to adjust AudioConfigSAMD21.h, which assumes 12 bits (setting the center point to 2048).

DavidHaitch, 2019-08-03:
I am experiencing a similar issue. Running the Sinewave demo on a Feather M0 produced no sound whatsoever, until I changed AUDIO_BIAS in AudioConfigSAMD21.h from 2048 to 1024. Then, I got output that sounded like @gregsadetsky 's "degraded" sinewave. 

@tfry-git: I would like to try and assist in testing fixes to this, but I don't know what you mean by scaling the audio output to different bit widths. I thought Mozzi always used 16 bits?

gregsadetsky, 2019-08-03:
Wow, it works!!!! Setting

`#define AUDIO_BIAS ((uint16_t) 512)`

works!! :-)

Thank you so, so much!

Should there be a pre-processor directive, or additional documentation about this?

DavidHaitch, 2019-08-08:
@gregsadetsky I am also working with a Feather M0, and I am unable to replicate your success. Even after setting AUDIO_BIAS to 512 in AudioConfigSAMD21.h, I still get a Sinewave that sounds like your degraded one.
Could you share the code that worked for you?

gregsadetsky, 2019-08-08:
I will take a look again, but I do remember the change being just what I posted above (the `#define` line).

Just to be sure (as this was a bit confusing to me initially), did you change the value in the `AudioConfigSAMD21.h` file that's within the Arduino library directory, or did you download a fresh copy of Mozzi that you included in your project directory, and made the change there?

I'm simply asking to make sure that you did the change in the code that was indeed compiled. Out of curiosity, if you delete the Mozzi directory where you did the change, does that create a compilation error?

Also, I remember trying much smaller values than 512 for fun and it did end up creating a "bit crushing" audio effect (a similar-ish distortion to what I posted). Could you try that (256, 128, etc.), just to make sure, again, that your changes in `AudioConfigSAMD21.h` are being taken into account? Cheers

tfry-git, 2019-08-09:
@DavidHaitch 
> I would like to try and assist in testing fixes to this, but I don't know what you mean by scaling the audio output to different bit widths. I thought Mozzi always used 16 bits?

Unfortunately, Mozzi does *not* always use 16 bits. For instance on the classic AVR architectures, it uses just over 8 bits, an STM32 it uses 10 bits, and on ESP8266 it uses 16 bits. You will always have to make sure that your updateAudio() returns the correct range of values for your configuration. This is not easy to fix, but see https://github.com/sensorium/Mozzi/issues/43 for one idea (and more background).

This might also be the reason why your sinewave still sounds distorted. Make sure updateAudio() returns values between -511 and 512.

@gregsadetsky 
> Should there be a pre-processor directive, or additional documentation about this?

Definitely. Do you think there is a compiler defintion that will identify the Feather M0, realiably? Then this could be used in AudoConfigSAMD21.h, as appropriate.

gregsadetsky, 2019-08-09:
@tfry-git 

> Definitely. Do you think there is a compiler defintion that will identify the Feather M0, realiably? Then this could be used in AudoConfigSAMD21.h, as appropriate.

There's an `ADAFRUIT_FEATHER_M0` flag/constant that - as far as I understand - comes from adafruit_samd's [boards.txt](https://github.com/adafruit/ArduinoCore-samd/blob/master/boards.txt#L46): `... extra_flags=... -DADAFRUIT_FEATHER_M0`.

[adafruit_samd](https://github.com/adafruit/ArduinoCore-samd) needs to be installed (see [here](https://learn.adafruit.com/adafruit-feather-m0-adalogger/using-with-arduino-ide#install-adafruit-samd-6-7)) in order to use the Feather M0 family of boards with the Arduino IDE.

As an example, you can see this constant used [here](https://github.com/adafruit/RadioHead/blob/master/examples/feather/RadioHead69_RawDemo_TX/RadioHead69_RawDemo_TX.ino#L26), for a Feather M0-based board with built-in radio ([this one](https://www.adafruit.com/product/3177)).

DavidHaitch, 2019-08-09:
@tfry-git I think I've found another possible issue.
[On line 518 of MozziGuts.cpp](https://github.com/sensorium/Mozzi/blob/master/MozziGuts.cpp#L518), the IS_SAMD21() case sets analogWriteResolution to 12 bits, even though the DAC on the SAMD21 is 10 bits.

[According to the analogWriteResolution docs](https://www.arduino.cc/reference/en/language/functions/zero-due-mkr-family/analogwriteresolution/), setting the resolution to something higher than the hardware's actual   resolution results in the remaining bits being dropped - only the first 10 are actually used. I don't have access to the hardware right now, but is that a possible problem?

tfry-git, 2019-09-30:
Yes, this is a possible problem. But it's really hard to fix this kind of problem, without having the hardware to test (and I don't that that, either).

tfry-git, 2019-11-04:
Hi!

Sorry for dropping the ball on this one. I have committed an experimental fix for this. Could you please do

```
git pull
git checkout testing/SAMD21_analog_resolution
```

and then test with that? If it works, I can merge this into the official branch. In theory this should now work out of the box (you'll still have to make sure updateAudio() returns values in the +/-512 range).

Thanks!

infovore, 2021-01-06:
I've been working with a SAMD21 and have run into this issue too. @tfry-git's solution in the branch is correct for _all_ SAMD21 chips.

The SAMD21 only has a single 10-bit DAC on it. (Its _ADC_s are 12-bit, however). This includes the Gemma M0 that support is listed for - that only has a 10-bit DAC output.

I'd put together my own PR but it seems a relatively straightforward fix...

tfry-git, 2021-01-06:
Thanks for the feedback. I have committed the following, then: https://github.com/sensorium/Mozzi/commit/f49e8d0db2b418d711a9a7c03e0bfb60274cb76e

I assume that closes the issue, but please reopen, if it doesn't.

closed 2021-01-06 by tfry-git

infovore, 2021-01-06:
Looks good - only other thing you might want to change is `AnalogWriteResolution` in `MozziGuts.cpp`, line 542. I didn't find this made any difference, but it's one thing I could think of.

infovore, 2021-01-07:
👍