sensorium/Mozzi #77: teensy 4.0 support?

CZHO404, 2019-09-03:
Well Teensy 4.0 is a thing and on the website for Mozzi it only lists Teensy 2.0 and 3.0. I'm a little confused if it will ever get support. Also i cant just try to use the 3.0 stuff because for some reason it only has A0 to A13.

PIN OUTS
Teensy2, 2++ | B5 (25)
Teensy3.x | A14 (analog out)
*link to pin out*
[https://www.pjrc.com/store/teensy40.html](url)

tfry-git, 2019-09-09:
Hi!

Adding Teensy 4.0 support will *probably* not be too hard. However, this will have to be done by somebody who has the hardware to test. Such as - you.

Main files to look at for starters:
- hardware_defines.h  - you'll probably have to add some define to IS_TEENSY3() (ignoring that this macro will then be a bit of a misnomer)
- AudioConfigTeensy3_12bit.h  - as you mention, you'll have to adjust the output pin

Chances are pretty good that this will be all you need to do. If it's not, check all occurances of IS_TEENSY3() in the Mozzi code for required adjustments.

If there are any questions: Ask. Oh, and don't forget to submit a PR, when done.

CZHO404, 2019-09-10:
Wow thank you for your time, ill try to get to that and have a nice day

closed 2019-09-10 by CZHO404

apiel, 2021-01-06:
@czho did you manage to get it work?

czho, 2021-01-06:
Sorry, I don't have the board and I have mostly lost interest in the project

czho, 2021-01-06:
No, sorry