sensorium/Mozzi #80: How to generate frequency lower then 1.5hz?

rahuldeo2047, 2019-10-28:
Ref https://github.com/sensorium/Mozzi/blob/a4ea518e75219c599e652d49795a6ec3a5114e27/Oscil.h#L186

I need to generate frequencies ranging 0.125hz to 25 Hz. For making a vibration table. And measure the vibration frequency and amplitude using an accelerometer fft. I will place the sensor just above the speaker or safest and nearest.

I can see the Mozzi can help me generate lowest frequency around 1.5hz. I need even lower. 

Q1. What are the best practices can help me achieve the given scenario with minimum noise?

I see this question relevant to Mozzi as it deals with the accuracy of frequency generated at given power/gain. As unit of frequency is Hz here ...

Q2. What is the unit of gain ( sound volume etc ) here?

Q3. Can we also check the SNR here?

Hope to contribute positively to this wonderful repository.

tfry-git, 2019-10-29:
Hi!

Mozzi can go lower than 1.5hz. In that function you refer to, you can go as low as 1/256 Hz, and with `setFreq_Q16n16()` you can go down to 1/65536Hz. However, you are right in suspecting that this is not the intended domain of Mozzi, and you will run into certain issues. Therefore it is a good idea to try to understand how the Oscil class works: In essence it is just a wave-table, and on each `updateAudio()` or rather, each `Oscil::next()` it will advance a fixed number of steps within the wave table. This number of steps is `phase_increment_fractional / 65536`.

Using an example `AUDIO_RATE` of 16384, a table size of 512 samples, and a target frequency of 1/8Hz, you'd arrive at `phase_increment_fractional` value of 64. That's still quite ok, as long as you don't expect to be able to produce very smooth frequency shifts, as the next frequency you will be able to hit is around 1.5% higher at ~ 0.127 Hz (phase_increment_fractional of 65). You can alleviate this by using a lower AUDIO_RATE, and a larger table size. If you are really desperate, you could even construct an Oscil with an increased `UPDATE_RATE` of e.g. `4*AUDIO_RATE`. This Oscil will then produce frequencies at 1/4 the advertised value.

TL;DR: Use a low AUDIO_RATE, use a large wavetable size, use `Oscil::setFreq_Q16n16()` for very low frequencies.

I'd also guess that using an appropriate low-pass filter (hardware) is a must in your use-case.

Q2: Oscil produces output values in the range -127 to +127, if that answers your question. And yes, that might be less than what you want. Replacing all ocurances of `int8_t` with `int16_t` inside Oscil.h (or better, creating a separate class with that adjustment) will be a quick-and-dirty way to up this by another 8 bits. Of course you'll also need a 16 bit wave_table, then, at a much larger number of samples. So your FLASH requirements have just grown, considerably, and probably beyond the classic Arduino boards. Also, be sure to use a high quality output method (e.g. HIFI_MODE). You see, improving this is not going to be trivial, but I'd recommend to experiment, first, and optimize later.

Q3: Sure, by recording and analysing the output. You'll be on your own, though. Remember to use that hardware low-pass filter.

tfry-git, 2019-10-29:
Afterthought on Q3: Your soundcard may or may not be suitable for recording, though. It may filter out the very low frequencies. In that case, I guess you'd have to resort to a USB oscilloscope for recording.

rahuldeo2047, 2019-10-29:
Thank you for your prompt and helpful response.
I'll study and experiment ( and keep seeking your help in going forward ) for my requirement.
I'll comeback.

closed 2021-01-08 by tfry-git