sensorium/Mozzi #81: External audio output

tomcombriat, 2019-11-02:
Hi!
As I was trying to port the MCP49XX DAC to Mozzi with an STM32 I found out that it was very hard to port different audio outputs to Mozzi as it usually has to deal with `MozziGuts.cpp`which is highly optimized and very hard to modify.

I was wondering if a good way to do this would be to allow the user to define a callback function, "extracting" the audio sample from Mozzi. I was thinking of a function like `void audioOutput(unsigned int sample)` in which the user could then send the sample to a DAC or anything else, and which would be defined in the user code, just like `updateControl` or `updateAudio` and be activated in `MozziConfig.h`.

Do you think it is worth that I try to implement such a thing (and remove the first port for an external DAC which would then be deprecated)? I think it won't affect the performances ofMozzi when using the standards outputs (PWM) and could be very useful for outputting the signal through other systems.

tfry-git, 2019-11-03:
Hi!

I think this is an excellent idea. I wonder (but am not sure) whether it would make sense to use this method, *always*. Would the compiler / linker be smart enough to optimize away the function call? If so, that would allow for an even cleaner / more consistent solution.

I suggest the existing DAC port would be kept, but moved to a dedicated include file, and implemented in terms of providing an ```audioOutput()``` function, too. If the user code includes that file, it's there, if it does not include it, it's not.

tomcombriat, 2019-11-03:
Hi again!

Alright cool! Thanks for your answer! I think this could simplify a lot the port to different devices while letting the user to define customized way of outputting the sound.

If you are okay with that, I will try in the next few days to implement that.
 In a first time I was thinking of removing the support for the external DAC from `MozziGuts.cpp` as it is not supported for other platforms than AVR (I am working on a port for STM32 which will be compatible with AVR) and implementing `void audioOutput(unsigned int sample)` and `void audioOutput2(unsigned int sample)` (for stereo). The "native" ways of outputting the sound would stay is `MozziGuts` for now, in order to give time to see if performances are remained when using these functions. 
The DAC support would come back in a second time, the time to code the new library for MCP49XX (or others) and using this new approach.

What do you think about this workflow?

tfry-git, 2019-11-03:
Sounds reasonable to me.

Some remarks:
1.  If we're making ```audioOutput()``` a part of the API, its signature should be well thought out. For one thing, it may or may not make sense to make it a 16 bit int, explicitly. (```unsigned int``` is 32 bits on ESP8266 and STM32, for instance).
2. Further, I think it would be least confusing, if the parameter to ```audioOutput()``` would remain 0-centered, i.e. a signed int. E.g. some DACs will accept signed input, anyway, some will many a biased unsigned. Thus any audio bias should be applied inside, not outside ```audioOutput()```. (This will mean re-visiting all calls to ```output_buffer.write()```, and ```output_buffer.read()```).
3. For the stereo-case, there should be one ```audioOutput(int l, int r)```, not two separate functions. This allows most flexibility for the output code (e.g. a stereo DAC will typically take left and right samples at once, not one after the other).

tomcombriat, 2019-11-03:
Alright,

1. I will think about it. I do not see why `int` type would not be suited, as ESP and STM will not gain anything when using a 16 bits function.
2. I was wondering about this. I agree that to be 0-centered would be more coherent but means that all the API (already implemented outputs methods) to be moved to this new paradigm, which (I do not know) could be a lot of work? 
3. Sure, completely agree with that.

tfry-git, 2019-11-03:
Regarding 1: Yes, I'm not sure what is the best answer, just something to think about.

Regarding 2: I hope it will not be a lot of trouble. The difficulty is that this is inconsistent across the output methods, already. *However*, you can easily find the mapping from zero-centered to biased by searching for ```AUDIO_BIAS``` in MozziGuts.cpp, and I believe reads and writes to the ```output_buffer``` are the only things that need adjusting. Also, you could delay this step, and simply remove ```AUDIO_BIAS``` if and when calling a user-defined ```audioOutput()```. Then we can worry about simplification / optimization, later.

tomcombriat, 2019-11-04:
Ok, I will try in the next few days to move everything to this new paradigm.
I will let you know how it goes. Thanks for your comments!

tomcombriat, 2019-12-24:
(I still have this project in mind, but did not have time to take care of it yet. As I am moving, I might take care of that in a couple of month, time for me to get stuff ready for tests. Sorry about that).

tomcombriat, 2020-05-17:
Closed by merge commit #87.

closed 2020-05-17 by tomcombriat