sensorium/Mozzi #82: Compile error Mozzi/mozzi_fixmath.cpp.o: In function `isqrt16(unsigned int)'

ouryperd, 2019-12-21:
I'm using the lib as part of a tutorial. I included it in Arduino Studio. I've looked around the lines that are reported, but I'm stumped. Compile error is:

Mozzi/mozzi_fixmath.cpp.o: In function `isqrt16(unsigned int)':
/home/temp/sketchbook/libraries/Mozzi/mozzi_fixmath.cpp:156: multiple definition of `isqrt32(unsigned long)'
echanter.cpp.o:/home/temp/sketchbook/libraries/Mozzi/Sample.h:61: first defined here
collect2: error: ld returned 1 exit status

tfry-git, 2019-12-21:
This error message does not immediately make sense.

1) What version of Mozzi is this? Is it unmodified?
2) Can you compile any of the examples shipped with Mozzi?
3) Can we see your code?

ouryperd, 2019-12-21:
It is the latest version.
I'm trying to do this: http://www.echanter.com/home/howto-build#TOC-Arduino-IDE
I'll try some of the examples and report back.

ouryperd, 2019-12-22:
I tried a few of the examples. They all compiled.

ouryperd, 2019-12-22:
I fixed it. In the root dir are two files with isqrt16(unsigned int) defined: mozzi_fixmath.cpp and mozzi_fixmath.h. I simply commented out the definition in the cpp file and it compiled.

tfry-git, 2020-01-17:
For what it's worth, the clash is not between the _declarations_ (multiple matching declarations are allowed), but between the _definitions_. There is one in mozzi_fixmath.cpp, and the other is in cogli_sqrti.h . Since cogli_sqrti.h already includes mozzi_fixmath.h, I think it should be safe (and cleanest) to remove the definition, there. In theory this should not break any code.

If there are no objections, I'll go ahead with that in a while.

closed 2023-04-10 by tfry-git