sensorium/Mozzi #83: Compile error for Arduino Nano Every board

davidspringer, 2020-01-16:
Hi guys,

I cannot get any Mozzi example sketches to compile when using an Arduino Nano Every board.

Is this a known compatibility issue? If so, is there a known workaround?

Thanks!

tfry-git, 2020-01-16:
The Nano Every is pretty fresh, and I for one have not laid my hands on one, yet. Supposedly the "Every" is compatible with the ATMega328 based boards, but I'm not sure whether or not that will also hold for the kind of low-level magic done in Mozzi. Either way, for a start, you could post the exact error message you are getting, then maybe we'll have a solution for that.

davidspringer, 2020-01-20:
Sure - from first glance it looks to be an incompatibility with the named registeres. Here are the first couple of lines, and attached is the full message:

C:\Users\Springer david\Documents\Arduino\libraries\Mozzi-master\mozzi_analog.cpp:43:3: error: 'ADCSRA' was not declared in this scope

   ADCSRA |= (1 << ADPS2);

   ^~~~~~

C:\Users\Springer david\Documents\Arduino\libraries\Mozzi-master\mozzi_analog.cpp:43:3: note: suggested alternative: 'ADC_t'

   ADCSRA |= (1 << ADPS2);

   ^~~~~~

   ADC_t

C:\Users\Springer david\Documents\Arduino\libraries\Mozzi-master\mozzi_analog.cpp:43:19: error: 'ADPS2' was not declared in this scope

   ADCSRA |= (1 << ADPS2);

                   ^~~~~
[Arduino Nano Every and Mozzi Error Messsages.txt](https://github.com/sensorium/Mozzi/files/4086513/Arduino.Nano.Every.and.Mozzi.Error.Messsages.txt)

tfry-git, 2020-01-20:
Hm, that doesn't look too good.

Supposedly, there is a compatibility mode that you can select among the board options. Is that enabled? If that doesn't help (or eats too many resources), I'm afraid it will have to implemented as an all new architecture for Mozzi. Often that isn't too hard for the more modern chips, which have many useful features already built in, but it will take a good deal of studying and testing.

davidspringer, 2020-01-21:
I've tried the "Registers Emulation: ATMEGA328" compatibility mode, and that doesn't seem to work, unfortunately.

I'll do some more digging to see if there are others who have managed to get some sort of compatibility to work.

diyelectromusic, 2021-12-15:
I was having a quick look at this too.  The "register emulation" only applies to the PORT and DDR registers and replaces direct register IO with C classes to translate over to the 4809's PORTs - so it is very slow!  It does nothing with other hardware registers, so all the places where timers are used (for example) will be wrong and need updating to work with the 4809's timer registers.

btvd, 2022-08-15:
Hey, guys! I've tried to install Mozzi on Arduino Every and also got those errors. Is there any workaround on how to use Ardino Nano Every with Mozzi?

tomcombriat, 2022-08-15:
Hi,
Never got any of these boards in my hands but I think the cleaner way to go would be to make a proper port of this arduino, but making an appropriate `MozziGuts_impl_NanoEvery.hpp` with all the correct values in there. Thanks to the template it should not be very hard but I cannot try without a board for testing. If someone wants to put its hands to it you are welcome to make a PR if you get it to work!

btvd, 2022-08-15:
Thomas, I got board but struggle on the procedure how to make MozziGuts_impl_NanoEvery.hpp. Can you guide me? Should I use the template in the root directory of this repo?

tomcombriat, 2022-08-15:
Yes, copy/paste the template and rename it to that, in the root folder. There are a lot of things to fill in but the instructions at the beginning of the file should be able to guide you at first. Looking at the implementations files for AVR and Teensy should also help in order to find out the correct functions. Basically it is *only* to find the correct timers functions and the correct way to output things fast. I think starting from the AVR template is probably the closest but unfortunately it is the hardest to read. As this platform is fairly recent, maybe starting from the Teensy implementation is a good idea.
Sorry, I am not sure if this is a lot of help :/…

diyelectromusic, 2022-08-15:
This was always on my "todo" list, but I've not got back to fiddling with Mozzi recently.  As indicated above, it is pretty much a different/new system and has very little resemblance to the other AVR versions.  And that reminds me, I still have the samd51 version too... to offer up in the new system (and I need to get up to date with the state of the RP2040 version...)

If you want a starting point for the Every, this might help work out the timers: https://emalliab.wordpress.com/2022/01/23/arduino-nano-every-timers-and-pwm/

Kevin

btvd, 2022-08-18:
Hey, all! Could you please also tell me what config file is related to Arduino Nano v3? I do not see implementation for ATmega328P, but lib works with them.

btvd, 2022-08-19:
Hey, guys! Can you please tell more on how to make AudioConfig specific to Nano Every. What should be covered there? I see that structure of this file is platform dependent.

tfry-git, 2022-08-20:
> Hey, guys! Can you please tell more on how to make AudioConfig specific to Nano Every. What should be covered there? I see that structure of this file is platform dependent.

At the minimum you should provide appropriate defines for AUDIO_BITS and AUDIO_BIAS. Also you will want to make the input configurable. See AudioConfigRP2040.h for a minimal example.

If there are other aspects of your implementation that users may want to tweak (e.g. the usage of hardware timers or other limited resources), AudioConfig is the place where to place defines for that.

diyelectromusic, 2022-11-03:
Ok, I'm going to take an initial look at this over the next week or so, unless someone else shouts that they are already looking into it...?

Progress (or not) will hopefully end up here: https://github.com/diyelectromusic/Mozzi/tree/megaAVRdev

Kevin

diyelectromusic, 2022-11-03:
> Progress (or not) will hopefully end up here: https://github.com/diyelectromusic/Mozzi/tree/megaAVRdev

Status update: Minimum viable noisemaking!

The basic structure is there for simple 8-bit resolution PWM output on D9.
Tested with Sinewave.  PWM frequency appears accurate.  440Hz Sinewave output appears accurate.
Tested with FMSynth.  Largely ok, but some crackling identified.  To be investigated.

No implementation for ADC, EXTERNAL_OUTPUT, or anything other than STANDARD_MODE for 8-bit, 16384Hz AUDIO_RATE.

Also, as all timers share a prescalar, to get the values required for 32768Hz PWM on D9 I have to change the system prescalar, so millis(), delay(), micros(), are now all running 64x too fast... but I don't see how that can be helped given how the ATmega4809's timers work and are implemented in the Arduino MegaAVR core...

So presented for any thoughts, comments, advice on resolution/etc.

Ideally it would be nice to mirror the "9bit" approach for the AVR core, but I can't follow through (without a lot more work) quite what is going on within the TimerOne utility library.  Like how come the internal PWM resolution is 488?  Does that come from the prescalar values used?  To be continued...

Kevin

diyelectromusic, 2022-11-03:
I think I might be able to support 9-bit PWM using single-slop PWM rather than dual-slope, as that gives me a counter TOP value of 488, mirroring (I think) what the original AVR port does for STANDARD and STANDARD_PLUS 9-bit modes...

Latest commit shows the difference between the two modes.  I don't know what difference it makes quality wise (I'm still getting crackles at present with the FM synth demo).

Also, this probably could support 32768Hz AUDIO_RATE but as this chip appears quite similar performance wise to the ATmega328, I'm guessing best would be to stick with a 32786Hz PWM and 16384Hz AUDIO_RATE... any opinions?

Kevin

diyelectromusic, 2022-11-08:
Ok, latest update:  There are two options for sound generation:
* Use D9 as output: this will mess with the system timing for millis(), delay(), etc, but gives AUDIO_RATE 16384Hz for PWM 32768Hz with "9-bit" resolution.  And of course D9 is the current default for the original Uno/Nano.
* Use D3 as output: this leaves the system timing alone, but only supports 8-bit PWM resolution.  It is still doing AUDIO_RATE 16384Hz, but requires a PWM rate of 65536Hz.

I think I'm still getting interference and crackles out of both modes though (but I can't rule out my kit, although I am using a lp filter/coupling cap on the output).  I'd be very interested in hearing from anyone interested in giving it a go to see what you think?

And if anyone is familiar with the use of the various audio/bit/resolution/timing settings in Mozzi ports more generally would like to check my working out, I'm still not totally sure I'm not getting something odd going on...

Aside: I don't think this is messing with the original AVR port, but I can't test for any other devices.

tomcombriat, 2022-11-17:
Hi!
Sorry for the late reply, I am also not the expert on AVR (I think it is more @sensorium ). Thanks for working on that port!

Just a few comments popping up while reading your last post:
 - I don't think you should worry screwing up `millis()` and all. If I am correct they are screwed up already in the existing AVR implementation and one should use mozzi's functions for that instead `mozziMillis()` et al. Using the same default pin also sounds like a good idea.
 - what kind of crackles do you get? Are they clearly interferences with something external or is it that the system cannot catch up with the sound generation (having the circular buffer running empty for instance). For instance, does that change when you change the `CONTROL_RATE`? Are very simple sketches like just a sine, affected?

diyelectromusic, 2022-11-21:
> * what kind of crackles do you get?

Sorry - I missed your reply!  I need to get back to this and have a proper look.  It might simply be the simple filter/amp set up I was using.  I don't think it is stuttering from a performance point of view, but I haven't done any measurements or anything on it to see for sure.  I think there might be something with the sine, that was my starting point for testing, but I've not experimented with the control_rate yet.

I guess I just wanted someone else to validate that this was a useful direction to explore, then I'll start testing and exploring a bit more thoroughly :)

Thanks,
Kevin