sensorium/Mozzi #85: AudioDelayFeeback example sketch won't compile for Arduino Uno

foglabs, 2020-03-15:
Hey, thanks for making Mozzi, its been great so far. I tried introducing `AudioDelayFeedback` into my project, and it wouldn't compile anymore. I finally tried compiling the example sketch for `AudioDelayFeedback`, and got the same result:

```
In file included from /Users/fog/Documents/Arduino/libraries/Mozzi/examples/09.Delays/AudioDelayFeedback/AudioDelayFeedback.ino:23:0:
/Users/fog/Documents/Arduino/libraries/Mozzi/AudioDelayFeedback.h: In member function 'void AudioDelayFeedback<NUM_BUFFER_SAMPLES, INTERP_TYPE>::write(int8_t, uint16_t)':
/Users/fog/Documents/Arduino/libraries/Mozzi/AudioDelayFeedback.h:176:50: error: lvalue required as left operand of assignment
   (write_pos + offset) &= (NUM_BUFFER_SAMPLES - 1);
                                                  ^
exit status 1
Error compiling for board Arduino Uno.

```

I noticed that this is the same situation as this old closed issue:
https://github.com/sensorium/Mozzi/issues/73

Let me know if I can provide any other details to help!

sensorium, 2020-03-16:
It compiles for Uno here.  Are you using the most recent code from Github?  What version of Arduino IDE are you using?

sensorium, 2020-03-16:
Does it help to add the cast to uint16_t in updateAudio?
  return asig/8 + aDel.next(asig, (uint16_t)deltime);

wyan, 2020-09-21:
I'm having the same problem compiling this sketch for the Leonardo. As I see it, the problem is that write_pos and offset are both `uint16_t`, so their sum is a number (an rvalue) instead of an expression that can be on the left of an assignment:
```
(write_pos + offset) &= (NUM_BUFFER_SAMPLES - 1);   // <<< Offending line
delay_array[write_pos] = input;
```

Should this be something like
```
write_pos &= (NUM_BUFFER_SAMPLES - 1);   // <<< Offending line
delay_array[write_pos+offset] = input;
```
instead?

sensorium, 2020-09-22:
Found the problem, thanks! Now:
    uint16_t _pos = (write_pos + offset) & (NUM_BUFFER_SAMPLES - 1);
    delay_array[_pos] = input;

Fixed on github, and the example also needed a change.  Let me know if it's still broken at your end.

wyan, 2020-09-23:
It now compiles fine on my side as well, I'll check it on the hardware tonight. Thanks a lot!

closed 2020-12-21 by sensorium