sensorium/Mozzi #86: Compile errors on Teensy 3.6 for Skeleton.ino with Arduino 1.8.11 and Teensy 1.51

wagnerluuv, 2020-03-22:
There seems to be some issue with the ADC.cpp library in MozziGuts and mozzi_analog

detailed description in forum link:

https://forum.pjrc.com/threads/59874-Teensyduino-1-51-errors-on-compile-1-48-does-not-ADC-class-implicated?p=233523&viewfull=1#post233523

wagnerluuv, 2020-03-22:
i have downgraded now to Arduino 1.8.10 and TeensyDuino 1.48 and it is able to compile... but i get these compile warnings:

`D:\LIBRARY\DOCUMENTS\Arduino\libraries\Mozzi-master\MozziGuts.cpp:45:2: warning: #warning "Mozzi has been tested with a cpu clock speed of 16MHz on Arduino and 48MHz on Teensy 3!  Results may vary with other speeds." [-Wcpp]

 #warning                                                                       \

  ^

D:\LIBRARY\DOCUMENTS\Arduino\libraries\Mozzi-master\mozzi_analog.cpp:140:13: warning: 'void adcSetChannel(uint8_t)' defined but not used [-Wunused-function]

 static void adcSetChannel(uint8_t channel) {

             ^

D:\LIBRARY\DOCUMENTS\Arduino\libraries\Mozzi-master\mozzi_rand.cpp:148:2: warning: #warning Automatic random seeding not implemented on this platform [-Wcpp]

 #warning Automatic random seeding not implemented on this platform

  ^

D:\LIBRARY\DOCUMENTS\Arduino\libraries\Mozzi-master\mozzi_utils.cpp: In function 'long int trailingZeros(long unsigned int)':

D:\LIBRARY\DOCUMENTS\Arduino\libraries\Mozzi-master\mozzi_utils.cpp:12:24: warning: dereferencing type-punned pointer will break strict-aliasing rules [-Wstrict-aliasing]

  return (*(uint32_t *)&f >> 23) - 0x7f;

                        ^

Sketch uses 18048 bytes (1%) of program storage space. Maximum is 1048576 bytes.
Global variables use 6148 bytes (2%) of dynamic memory, leaving 255996 bytes for local variables. Maximum is 262144 bytes.`

wagnerluuv, 2020-03-22:
while I'm here, I'll just ask another question, with which someone could might help me:

I can't find the A14 output on the Teensy 3.6

Can someone point out to me, which pin to use on this image:
https://www.pjrc.com/store/teensy36b.jpg

sensorium, 2020-03-24:
https://www.pjrc.com/teensy/pinout.html