sensorium/Mozzi #88: twi_nonblock not in master branch

moriczgergo, 2020-04-20:
I had to download the release instead. Is there a reason for this?

tfry-git, 2020-04-20:
Apparently it was deleted in commit 584489fc33637c15cf315ec525582a8cb2c22741 . Probably a mistake?

sensorium, 2020-04-27:
Removed by accident.  Now replaced.  I've been experimenting with more general nonblocking Wire and twi code for use with Mozzi, but I'm not sure it would simplify things much for users.  It still requires others to understand my modifications to the Wire code which might take as long as working out their own version for use with their particular I2C hardware.

moriczgergo, 2020-05-09:
Alright. What could be good is a fake Wire object to pass to drivers, like most Adafruit OLED drives.

closed 2020-05-09 by moriczgergo