sensorium/Mozzi #89: Error compiling on linux with stm32 all boards

CZHO404, 2020-04-23:
Lately, I have been experimenting with stm32 boards and I was very happy to see that stm32 was supported here. The bad thing was my computer for some reason doesn't compile right, the error is as follows: 

/home/czho/Arduino/libraries/Mozzi-master/MozziGuts.cpp:34:10: fatal error: STM32ADC.h: No such file or directory
   34 | #include <STM32ADC.h>
      |          ^~~~~~~~~~~~
compilation terminated.
exit status 1
Error compiling for board Generic STM32F1 series.

This happens on all boards and I'm very confused.

tfry-git, 2020-04-24:
Hi!

Confusingly there are two different STM32 arduino cores out there. The one that Mozzi has been ported to is this: https://github.com/rogerclarkmelbourne/Arduino_STM32 . I assume you are probably using the other.

CZHO404, 2020-04-24:
Thanks for the help, I heard that that one was better and has more libraries and stuff. In fact, the USB midi thing I was using was only compatible with that one. I had some issues with installing that one on Linux tho so ill just manage.

closed 2020-04-24 by CZHO404

CZHO404, 2020-04-24:
Compiles, works but does not work on stm32f4, I assume it is not supported but could be easily implemented, it says that #include <STM32ADC.h> No such file or directory. I changed all the <STM32ADC.h> to <STM32F4ADC.h> which you could just make an #if statement for. Then all you would have to do is change the CPU speed, the only reason I can't do this is that I'm not really acquainted with this stuff and don't know how to.

tfry-git, 2020-04-25:
Problem is, I don't have an f4 board to test with. Are you sure there are CPU issues? On STM32 Mozzi isn't quite so close to the bare metal, and so I would actually expect this to work. And if you're curious, the code starting at this line (https://github.com/sensorium/Mozzi/blob/6b1a1270c027cb487262131395c18911db6eae36/MozziGuts.cpp#L521) sets up the timers for STM32, and is relatively harmless, IMO. If there are any specific questions, just ask.

Maybe you would like to give it a try yourself, and create a pull request for whatever needs to be done?

CZHO404, 2020-04-25:
Hi, there actually is a lot of errors involving the cpu speed: 

> /home/czho/Arduino/libraries/Mozzi-master/MozziGuts.cpp:546:2: error: #error Configured audio resolution is definitely too high at the configured audio rate (and the given CPU speed)
>  #error Configured audio resolution is definitely too high at the configured audio rate (and the given CPU speed)
>   ^
> /home/czho/Arduino/libraries/Mozzi-master/MozziGuts.cpp: In function 'void startAudioStandard()':
> /home/czho/Arduino/libraries/Mozzi-master/MozziGuts.cpp:525:25: error: 'F_CPU' was not declared in this scope
>    uint32_t period_cyc = F_CPU / AUDIO_RATE;
>                          ^
> /home/czho/Arduino/libraries/Mozzi-master/MozziGuts.cpp:530:22: error: 'class HardwareTimer' has no member named 'setChannel1Mode'
>    audio_update_timer.setChannel1Mode(TIMER_OUTPUT_COMPARE);
>                       ^
> /home/czho/Arduino/libraries/Mozzi-master/MozziGuts.cpp:533:22: error: 'class HardwareTimer' has no member named 'attachCompare1Interrupt'
>    audio_update_timer.attachCompare1Interrupt(pwmAudioOutput);
>                       ^
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp: In function 'void setupFastAnalogRead(int8_t)':
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp:57:32: error: 'class STM32ADC' has no member named 'setSampleRate'
>   if (speed == FASTEST_ADC) adc.setSampleRate(ADC_SMPR_1_5);
>                                 ^
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp:57:46: error: 'ADC_SMPR_1_5' was not declared in this scope
>   if (speed == FASTEST_ADC) adc.setSampleRate(ADC_SMPR_1_5);
>                                               ^
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp:58:36: error: 'class STM32ADC' has no member named 'setSampleRate'
>   else if (speed == FASTER_ADC) adc.setSampleRate(ADC_SMPR_13_5);
>                                     ^
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp:58:50: error: 'ADC_SMPR_13_5' was not declared in this scope
>   else if (speed == FASTER_ADC) adc.setSampleRate(ADC_SMPR_13_5);
>                                                   ^
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp:59:19: error: 'class STM32ADC' has no member named 'setSampleRate'
>          else (adc.setSampleRate(ADC_SMPR_41_5));
>                    ^
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp:59:33: error: 'ADC_SMPR_41_5' was not declared in this scope
>          else (adc.setSampleRate(ADC_SMPR_41_5));
>                                  ^
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp: In function 'void setupMozziADC(int8_t)':
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp:76:6: error: 'class STM32ADC' has no member named 'calibrate'
>   adc.calibrate();
>       ^
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp: At global scope:
> /home/czho/Arduino/libraries/Mozzi-master/mozzi_analog.cpp:140:13: warning: 'void adcSetChannel(uint8_t)' defined but not used [-Wunused-function]
>  static void adcSetChannel(uint8_t channel) {
>              ^
most of them boiling down to the config speed being wrong, as I said im not experienced with this lib and idk how I would even try to get the errors fixed, so Im just going to switch to stm32f1 cause of the fact that theres way more support for it. I might get to it in the future, but thanks for the help, I was compleatly stuck, I was so frustrated cause most of the guides that told me how to use this board didnt even mention that core by rogerclarkmelbourne.