sensorium/Mozzi #92: Is ther an Audio Buffer in Mozzi?

jtragtenberg, 2020-07-08:
Hello I lloked through the Mozzi library code and I couldn’t find where I could define the Buffer size.
Does it work with a buffer with a predefined number of audio samples? Or is processing just done in the time between one sample and another?

sensorium, 2020-07-08:
Hi,
the audio buffer is an instance of the class from CircularBuffer.h, called "output_buffer" in MozziGuts.cpp.  The CircularBuffer class has a hard-coded size of 256, for execution speed.  You'll need to code the class differently if you change the size, as it uses some bit arithmetic for advancing the pointers.  There's not much to it, so it won't be hard to experiment.

jtragtenberg, 2020-07-08:
So the buffer has the size of one sample. Was there a reason for that choice? If there was a buffer with 16 audio samples, instead of having 61 microseconds to process each sample, we could have 976us or even 1,9ms using a buffer with 32 audio samples. It would take longer to fill and read the buffer, the amount of time to process the signal would increase dramatically, wouldn't it? It would even be possible to make more elaborate processing with a 32768 Hz sample rate.
To implement it I believe it would be better to have another circular buffer holding the last N item[] arrays from the current circularBuffer, but I don't understand much about optimising C code. What do you think about it?

sensorium, 2020-07-09:
The buffer has a size of 256 audio samples, giving about 15ms latency at 16384Hz.  You can look at the code in CircularBuffer.h.

jtragtenberg, 2020-07-09:
Oh, that is good to hear, thank you.

closed 2020-07-09 by jtragtenberg