sensorium/Mozzi #93: Arduino m0 and m0 pro compiles with no errors, was wondering if it is officially supported.

CZHO404, 2020-07-25:
So I was making a synth with the STM32 boards but I just realized how the Arduino m0 and m0 pro boards are basically the same things. The gemma m0 is officially supported and is basically the same thing as the Arduino m0, so it might be a mistake that it is not shown as supported, it compiles with no errors. They are both atsamd21 board with the Arduino m0 being ATSAMD21G18 and the gemma m0 being ATSAMD21E18.

Thank you.

tfry-git, 2020-10-09:
The Mozzi developers simply don't have access to all that many different boards. If it compiles for you that is a good hint that it will, in fact, work. Since presumably you own an Arduino zero, why don't you just give it a try, and let us know about the result?

CZHO404, 2020-10-09:
well you see I have the same problem as the mozzi devs, I was trying out boards for a project and didnt have access to an arduino zero but while I was looking thru the code of mozzi it had implemented atsamd21 support not just gemma m0. With the chips being so similar I assumed it was but got confused when nothing was listed about it.