sensorium/Mozzi #100: Joining Forum? Not a bug....

poetaster, 2020-11-30:
Hi, I'm trying to join the forum, wanting to share the Mozzi build I've made available. I don't seem to be able to get in? 

Thanks for the great Library!

My Mozzi project is at:
https://github.com/poetaster/rampart

sensorium, 2020-11-30:
Hi Mark,

what happens when you try to join the group?  Normally I get an email saying someone wants to join, and I tick a box to say yes, as a way to filter spammers.  I didn't see any requests recently - do you want to try again?  It would be great if you are able to share your project...

Tim

poetaster, 2020-11-30:
> Hi Mark,

Hi!

> 
> what happens when you try to join the group?  Normally I get an email
> saying someone wants to join, and I tick a box to say yes, as a way
> to filter spammers.  I didn't see any requests recently - do you want
> to try again?  It would be great if you are able to share your
> project...

Ok, I see 'Join request Sent' and am unable to do anything else... 

Thanks for Mozzi! and
for replying so quickly!

Mark

> 
> Tim
> 
> > 
> > 
> > Hi, I'm trying to join the forum, wanting to share the Mozzi build
> > I've made available. I don't seem to be able to get in?
> > 
> > Thanks for the great Library!
> > 
> > My Mozzi project is at:
> > https://github.com/poetaster/rampart
> > <https://github.com/poetaster/rampart> —

closed 2022-12-06 by poetaster