sensorium/Mozzi #102: //#define EXTERNAL_AUDIO_OUTPUT true removed from mozzi_config.h?

tomcombriat, 2020-12-27:
Hi!

I have seen that, in the last merge, the option in `mozzi_config.h` for allowing the user to define an external audio output function has been removed. Unless I have missed something, this change removes all DAC capabilities of Mozzi that are in the examples/External_Audio_Output folder. 

Is that a mistake?

tfry-git, 2020-12-27:
Hi!

Sorry, my bad. This wasn't intentional.

Actually I do plan on breaking the current external audio output option (in a way that will be easy to adjust to). See https://github.com/sensorium/Mozzi/pull/98 . This PR (among other restructuring) also moves the other platforms' audio output code to AudioOutput.h, and adds some more options to customize buffering. The - slightly - incompatible change, in here, is that stereo samples are handled in a simple struct. That little change allows several simplifications inside MozziGuts.cpp, and so, I think, is worth it.

In any case, feedback on that PR will be very welcome.

Thomas

tomcombriat, 2020-12-27:
Ok!

Actually, I was working on a new port (PT8211 on the STM32), that was compatible with the previous external audio output function. Is that worth that I pull request it?

closed 2020-12-28 by tomcombriat