sensorium/Mozzi #105: Error on compile Mozzi STM32F boards

pirvcb, 2021-01-10:
Hello there. I'm not a trained coder so i can't investigate in deep. I just LOVE MOZZI LIBRARY!
Anyway, any try to compile for STM32 board gives me the same error.
Other NOT MOZZI sketches compiling OK.
Libraries are:
-Mozzi Master 
- STM32 Master roger clark melbourne

Tried on both Mac and PC. Same result.

TIA!
Davide

Arduino: 1.8.13 (Mac OS X), Board: "Generic STM32F103C series, STM32F103C8 (20k RAM. 64k Flash), STLink, 72Mhz (Normal), Smallest (default)"

/Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/arduino-builder -dump-prefs -logger=machine -hardware /Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/hardware -hardware /Users/asinus/Library/Arduino15/packages -hardware /Users/asinus/Documents/Arduino2/hardware -tools /Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/tools-builder -tools /Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/hardware/tools/avr -tools /Users/asinus/Library/Arduino15/packages -built-in-libraries /Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/libraries -libraries /Users/asinus/Documents/Arduino2/libraries -fqbn=Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=STLinkMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10813 -build-path /var/folders/18/lmbggqdj6nscfzk_cb79d7dw0000gq/T/arduino_build_50835 -warnings=none -build-cache /var/folders/18/lmbggqdj6nscfzk_cb79d7dw0000gq/T/arduino_cache_666148 -prefs=build.warn_data_percentage=75 -verbose /Users/asinus/Documents/Arduino/libraries/Mozzi/examples/01.Basics/Sinewave/Sinewave.ino
/Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/arduino-builder -compile -logger=machine -hardware /Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/hardware -hardware /Users/asinus/Library/Arduino15/packages -hardware /Users/asinus/Documents/Arduino2/hardware -tools /Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/tools-builder -tools /Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/hardware/tools/avr -tools /Users/asinus/Library/Arduino15/packages -built-in-libraries /Users/asinus/Desktop/Arduino 1.8.13.app/Contents/Java/libraries -libraries /Users/asinus/Documents/Arduino2/libraries -fqbn=Arduino_STM32-master:STM32F1:genericSTM32F103C:device_variant=STM32F103C8,upload_method=STLinkMethod,cpu_speed=speed_72mhz,opt=osstd -ide-version=10813 -build-path /var/folders/18/lmbggqdj6nscfzk_cb79d7dw0000gq/T/arduino_build_50835 -warnings=none -build-cache /var/folders/18/lmbggqdj6nscfzk_cb79d7dw0000gq/T/arduino_cache_666148 -prefs=build.warn_data_percentage=75 -verbose /Users/asinus/Documents/Arduino/libraries/Mozzi/examples/01.Basics/Sinewave/Sinewave.ino
Using board 'genericSTM32F103C' from platform in folder: /Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1
Using core 'maple' from platform in folder: /Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1
Detecting libraries used...
/Users/asinus/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10813 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/include -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/stm32f1/include -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/stm32f1 -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/usb_lib -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/cores/maple -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/variants/generic_stm32f103c /var/folders/18/lmbggqdj6nscfzk_cb79d7dw0000gq/T/arduino_build_50835/sketch/Sinewave.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
Alternatives for MozziGuts.h: [Mozzi-master@1.1.0]
ResolveLibrary(MozziGuts.h)
  -> candidates: [Mozzi-master@1.1.0]
/Users/asinus/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10813 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/include -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/stm32f1/include -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/stm32f1 -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/usb_lib -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/cores/maple -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/variants/generic_stm32f103c -I/Users/asinus/Documents/Arduino2/libraries/Mozzi-master /var/folders/18/lmbggqdj6nscfzk_cb79d7dw0000gq/T/arduino_build_50835/sketch/Sinewave.ino.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/Users/asinus/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10813 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/include -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/stm32f1/include -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/stm32f1 -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple/usb/usb_lib -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/cores/maple -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/variants/generic_stm32f103c -I/Users/asinus/Documents/Arduino2/libraries/Mozzi-master -I/Users/asinus/Documents/Arduino2/libraries/Mozzi-master/utility /Users/asinus/Documents/Arduino2/libraries/Mozzi-master/utility/FrequencyTimer2.cpp -o /dev/null -DARDUINO_LIB_DISCOVERY_PHASE
/Users/asinus/Library/Arduino15/packages/arduino/tools/arm-none-eabi-gcc/4.8.3-2014q1/bin/arm-none-eabi-g++ -c -g -Os -w -DDEBUG_LEVEL=DEBUG_NONE -std=gnu++11 -ffunction-sections -fdata-sections -nostdlib --param max-inline-insns-single=500 -fno-rtti -fno-exceptions -fno-use-cxa-atexit -DBOARD_generic_stm32f103c -DVECT_TAB_ADDR=0x8000000 -DERROR_LED_PORT=GPIOC -DERROR_LED_PIN=13 -w -x c++ -E -CC -mcpu=cortex-m3 -DF_CPU=72000000L -DARDUINO=10813 -DARDUINO_GENERIC_STM32F103C -DARDUINO_ARCH_STM32F1 -DCONFIG_MAPLE_MINI_NO_DISABLE_DEBUG=1 -DSERIAL_USB -DGENERIC_BOOTLOADER -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -DMCU_STM32F103C8 -mthumb -march=armv7-m -D__STM32F1__ -I/Users/asinus/Documents/Arduino2/hardware/Arduino_STM32-master/STM32F1/system/libmaple -I/Users/asinus/Docume [... cut ...]
/Users/asinus/Documents/Arduino2/libraries/Mozzi-master/mozzi_analog.cpp:24:22: fatal error: STM32ADC.h: No such file or directory
 #include <STM32ADC.h>
                      ^
compilation terminated.
/Users/asinus/Documents/Arduino2/libraries/Mozzi-master/MozziGuts.cpp:36:22: fatal error: STM32ADC.h: No such file or directory
 #include <STM32ADC.h>
                      ^
compilation terminated.
Using library Mozzi-master at version 1.1.0 in folder: /Users/asinus/Documents/Arduino2/libraries/Mozzi-master 
exit status 1
Error compiling for board Generic STM32F103C series.

tomcombriat, 2021-01-11:
I had the same problem, there are a few files, including STM32ADC.h which need to be copy pasted from your STM32 installation. There are not there by default as they prevent mozzi from compiling for AVR (or maybe I did a mistake somewhere while installing, in that case you are in the same state than me). A simple search in the Arduino installation folder should help you to find them. If I remember well, there are also a few that goes in the Mozzi/utility folder…

Hope it helps,

tfry-git, 2021-01-11:
For some obscure reason, placing the ```#include <STM32ADC.h>``` in AudioConfigSTM32.h instead of MozziGuts.cpp fixes this issue. I have mixed that into my PR (https://github.com/sensorium/Mozzi/pull/98), as I stumbled into the issue while testing compilation of my changes for STM32.

In summary: The cause is unknown (a bug in the Arduino enviroment, I'll say), the solution is known, but the fix will have to wait a while, as I don't want to create a merge conflict.

pirvcb, 2021-01-12:
Thanks Thomas and Tom!!!
I tried to move #include <STM32ADC.h> in the right file as Thomas said but no luck.
Again errors, but different errors this time.
I linked a folder where are log and modified files from my arduino installation.
Before to try again in "Blind and Casual Mode", i'll wait for some signal from you, if this would be possible.
The simplest solution should be uploading the already modified Mozzi folder to let me sobstitute it inside my Arduino installation!
Thanks again for your time and patient!
Davide

https://drive.google.com/drive/folders/1sxKroTjK-fDSgsWZ-rHIB_xk_RYFLI_E?usp=sharing

tomcombriat, 2021-01-12:
Hey,
There should be more files to move (at least here) to get it work. Apparently Thomas has found a long term fix, but in the meantime if you want to try, [this repo](https://github.com/tomcombriat/Mozzi/tree/TES-stable) is the fork I'm using to compile on the STM32F boards, with some files for STM added in the library folder. It's possibly not completely up to date, but it should be close, and could worth a try.

Hope it helps!

pirvcb, 2021-01-12:
> Hey,
> There should be more files to move (at least here) to get it work. Apparently Thomas has found a long term fix, but in the meantime if you want to try, [this [...]

Thank you Thomas.
Replaced Mozzi folder with your suggested Mozzi TES and compilation goes at end. I just got the following Warning:

**libraries\Mozzi-TES-stable\MozziGuts.cpp.o: In function `pwmAudioOutput':
C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable/MozziGuts.cpp:479: warning: undefined reference to `audioOutput(int, int)'**

Now i try to upload in real hardware and let see....
Thanks again for your kind support!!!
Davide

pirvcb, 2021-01-12:
Uploaded firmware but no sounds from PB8. Changed any possible audio config (PWM RATE and/or HIFI mode) but nothing.
I'm still waiting for a Maple board. I just tried a Blue Pills 3 buttons usb C  WeActTC 64k STMF103C8T6 chinese board.
I'll give you another report when Maple will arrive here.
Thanks again!
AVR forever????
BAH!
Later...

tomcombriat, 2021-01-13:
Hi,

Given the number of the line, the problem seems to be at line ` audioOutput((int) output_buffer.read());`.
If you have configured mozzi to not use the `STEREO_HACK`, nor `EXTERNAL_AUDIO_OUTPUT` in mozzi_config.h this should be working. The line you referred to should not be compiled. 

Are you using Clack's version of STM32duino?

pirvcb, 2021-01-13:
Hi!
Yes, i use STM32 Clark latest version.

This Blue Pill board seems in other ways working ok. Tried the blinking led and it works.

Dunno, i'm going to stop myself here, for now, going back to AVR.
Waiting for Maple delivery...
Thanks again for all Thomas!!!
As soon as i'll try again i'll give you some feedback.
Cheers
Davide

pirvcb, 2021-01-19:
Hi Thomas. 
I'm back.
Got the Maple. But for an accident i have to uninstall Arduino.
So: Uninstalled and reinstalled the whole arduino IDE.
Put in right places RogerClarck's folder and your Mozzi TES stable.
And now.....
WHAT IS THIS?

In file included from C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable\MozziGuts.cpp:24:0:
C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable\AudioOutput.h:79:2: warning: #warning "Mozzi is configured to use an external void 'audioOutput(int l, int r)' function. Please define one in your sketch" [-Wcpp]
 #warning "Mozzi is configured to use an external void 'audioOutput(int l, int r)' function. Please define one in your sketch"
  ^
C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable\MozziGuts.cpp:533:2: warning: #warning Configured audio resolution may be higher than optimal at the configured audio rate (and the given CPU speed) [-Wcpp]
 #warning Configured audio resolution may be higher than optimal at the configured audio rate (and the given CPU speed)
  ^
In file included from C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable\MozziGuts.cpp:24:0:
C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable\AudioOutput.h:9:13: warning: inline function 'void audioOutput(int, int)' used but never defined [enabled by default]
 inline void audioOutput(int, int =0);
             ^
C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable\utility\util_adc.c:10:30: warning: 'adc_result' initialized and declared 'extern' [enabled by default]
 extern volatile unsigned int adc_result = 0;
                              ^
libraries\Mozzi-TES-stable\MozziGuts.cpp.o: In function `pwmAudioOutput':
C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable/MozziGuts.cpp:479: warning: undefined reference to `audioOutput(int, int)'
Lo sketch usa 17096 byte (26%) dello spazio disponibile per i programmi. Il massimo è 65536 byte.
Le variabili globali usano 4320 byte (21%) di memoria dinamica, lasciando altri 16160 byte liberi per le variabili locali. Il massimo è 20480 byte.

I know, this is not exactly my core job in mozzi. Anyway, i'd like have a fast platform working with your Mozzi porting.
TIA
Davide

tomcombriat, 2021-01-19:
Hi Davide,

It is likely that my fork is now configured to use an external audio output function (see PR #87), this allows to use external dac chips with Mozzi, which is what I do mainly. You can disable it in mozzi_config.h (probably a `#define EXTERNAL_AUDIO_OUTPUT` that you need to comment in this file. This will likely solve a lot of lot of the compiling error.

Seems also that you modified the audio resolution, but you might have done that on top of my fork…

Hope it helps,

pirvcb, 2021-01-19:
Thomas,
as you told me, commenting out this line "" //#define EXTERNAL_AUDIO_OUTPUT false  "" in mozzi_config.h solves compiling errors. AudioRate also, as you right pointed, needs to be set to 32768 and AudioMode needs set to HIFI_Mode.
In this case no more warning.

I just would like ask you if there is a way to set AusioMode to StandardPlus without errors. (?)

Anyway, not yet tested in hardware but i'll try in few hours.
Thanks again.

tomcombriat, 2021-01-19:
> In this case no more warning.

I always get warning when compiling, some of them are coming from STM32duino, some from other librairies, and a lot should not be considered a problem if sound is coming from the hardware.

> BUT NOW I GOT THIS: […]

Same, probably no need to worry about.

> I need to use internal DAC!!!

If you comment the line I was talking about, then the sound will be outputted on the BP pin. Note that it is not a proper DAC, but PWM.

> AudioMode needs set to HIFI_Mode.

In that mode, sound will be outputted on two pins (see Mozzi documentation on that). To begin with, I would suggest to try to get something with the STANDARD_PLUS mode.

pirvcb, 2021-01-19:
AudioMode needs set to HIFI_Mode.

In that mode, sound will be outputted on two pins (see Mozzi documentation on that). To begin with, I would suggest to try to get something with the STANDARD_PLUS mode as a starter ;)

Is exaclty thai i would do, try in standard plus i mean. But if i change the "default" HIFIMODE to Standard Plus i got this:

C:\Users\kkk\Documents\Arduino\libraries\Mozzi-TES-stable\MozziGuts.cpp:533:2: warning: #warning Configured audio resolution may be higher than optimal at the configured audio rate (and the given CPU speed) [-Wcpp]
 #warning Configured audio resolution may be higher than optimal at the configured audio rate (and the given CPU speed)

tomcombriat, 2021-01-19:
This might work but not optimal… 

My guess is that you have changed something in `AudioConfigSTM32.h`. Line 23 should be: `#define AUDIO_BITS 10`, as it is by default. 

Honestly, it has to be tested on the hardware, with loudspeaker or a scope connected. I usually overlook all the warnings given by the compiler at first and then, sometimes, if something goes not as intended, take a closer look at them.

This seems not far from working ;)

pirvcb, 2021-01-19:
HI THOMAS!
Got it working finally!
Node was commenting out this linein mozzi_config.h  //#define EXTERNAL_AUDIO_OUTPUT false
as you stated.
 Line 23 was as default: #define AUDIO_BITS 10; no modified by me.
The other compiler warning audio rate related, seems strange, but after the first time compilation (for any new sketche charged), at second try they gone..... Magic world!

Anyway, real hardware as at start described, Blue Pill USB C type with buttons and no jumpers, works!
Thank you again and again for your time and your kindly patiente!
Davide

tomcombriat, 2021-01-20:
Hi,

Glad to hear it! The fact that the warnings do not show up anymore is because not everything is recompiled each time… Not very important though…

If it is working, maybe you could mark this issue as "closed" in order to clear a bit this list.

Have good fun with Mozzi!

closed 2021-01-20 by pirvcb