sensorium/Mozzi #110: STM32 Audio Delays examples

pirvcb, 2021-01-24:
Hello there.
In STM32, TES STABLE Thomas Combriat library, compiling Audio Delay related examples, gives out a non working skectch.
Audio is there but not audible delay.
Tried to change variable type (in AudioDelayFeedbackX2 stops to compile till you change in "unsigned long" these:
unsigned long del_samps1;
unsigned long del_samps2;) but no rezsults in audio output.
Changed samples cells and nothing happened.
Somebody else stumbled in this ?

tfry-git, 2021-01-24:
Not tested, but I think you need to declare these as uint16_t, instead, in order to trigger the correct overload. Could you give that a try?

pirvcb, 2021-01-24:
Later i'll try and report here.

pirvcb, 2021-01-24:
On example AudioDelayFeedbackX2, declaring  del_sampels as uint16_t , solves the problem as you stated.

In AudioDelayFeedback, (not X2), things are working ok with byte variable in the first two examples:
int updateAudio(){

  char asig = aTriangle.next(); // get this so it can be used twice without calling next() again
  **_//return asig/8 + aDel.next(asig, (uint16_t) del_samps); // mix some straight signal with the delayed signal
  //return aDel.next(aTriangle.next(), (uint16_t) del_samps); // instead of the previous 2 lines for only the delayed signal_**

third version needs del_samps declared as uint16_t otherwise no soud out at all.
But, anyway, it still dont gives out any delay. Just the carrier.
 **_return asig/8 + aDel.next(asig, (uint16_t) deltime); // mix some straight signal with the delayed signal_**
}

closed 2021-01-25 by pirvcb