sensorium/Mozzi #115: Compile on PC (Linux, ALSA, SDL2)?

Harvie, 2021-01-31:
Hello,
i would like to compile Mozzi on PC (Linux) to be able to simulate its output and test synth designs without having to run it on actual hardware. Have you been thinking about adding support for such layer?

I was thinking about using SDL2 multiplatform library (or similar) to output the audio as it would be possible to compile with little to none modifications on Linux, Windows, MacOS and BSD.

Do you think this is possible?

tfry-git, 2021-02-01:
Hi,

> Have you been thinking about adding support for such layer?

No, but I think we'll be happy to accept a patch.

> Do you think this is possible?

This should be possible, but not quite sure, how easily. First you'll need a PC core for Arduino. This one may or may not be suitable: https://github.com/maniacbug/ncore . For the actual adaptations needed inside Mozzi, take a look at MozziGuts.cpp and AudioOutput.h (recent git checkout, as this has been refactored quite a bit for better readability).

Harvie, 2021-02-01:
> First you'll need a PC core for Arduino.

I would much rather cut Arduino specific code out using ifdefs altogether. Then use Makefile to build instead (doesn't mean Arduino IDE support is not possible, but it does not really make sense to me). What do you really use from Arduino core except for the hardware specific code? Right now my build failed on missing WProgram.h

> No, but I think we'll be happy to accept a patch.

I kinda started doing some modifications, but i don't really know how Mozzi works internaly. For example how can i specify sampling frequency of Mozzi. Because while on Atmega you want to keep sampling frequency to something that you can handle on time, on PC it makes more sense to use sampling frequency natively supported by soundcard to prevent resampling.

tfry-git, 2021-02-01:
> I would much rather cut Arduino specific code out using ifdefs altogether.

Sure, but as Mozzi is designed for MCUs and the Arduino environment, that may actually be the much harder route. Importantly, Mozzi is already ported to different Arduino cores, and thus you can follow that approach by example. As to what is actually used from the Arduino core: Good question, but you'll probably have to find out by yourself.

> For example how can i specify sampling frequency of Mozzi

The AUDIO_RATE parameter in mozzi_config.h.

For a rough overview of Mozzi's internal workings, perhaps the write-up on top of AudioOutput.h will be of some help.

Harvie, 2021-02-01:
> The AUDIO_RATE parameter in mozzi_config.h.

It says following:

 ```
Holds the audio rate setting.                                                                                                           
 AUDIO_RATE can be \#defined as 16384 or 32768 Hertz in Mozzi/mozzi_config.h.
```

Would i be able to set any arbitrary rate when not running on micro?

tfry-git, 2021-02-01:
I believe so, but non-powers of two have not seen too much testing, for all I know.

apiel, 2021-02-20:
It would also be cool to have this lib working on raspberry, for example on the rpi zero that doesn't have sound output, using a pwm pin would be great.

Harvie, 2021-02-20:
Actually the rpi zero pwm audio output can be enabled and will work out of box with any application capable of playing audio on linux.
So if we will have ALSA or SDL2 support in Mozzi, it will be possible to use pwm audio on RPI zero without any modifications.

pschatzmann, 2021-05-07:
Hallo Harvie,  Did you advance on this and do you still need some help?

pschatzmann, 2021-07-04:
I am currently working an a redesign of the output functions to simplify the implementation of new processor platforms.
Quite some time ago I also implemented an Arduino Emulator for Linux. 
With this your requirement was quite easy to implement. 

You can check out the LinuxSupport feature branch of https://github.com/pschatzmann/Mozzi
Then you just need to build the cmake project e.g. with
 
- mkdir build
- cd build
- cmake -DBUILD_EXAMPLE=ON ..
- make

This builds the MOZZI executable. 

However I was using PulseAudio instead of Also because I wanted the solution to run on OS/X as well....