sensorium/Mozzi #116: LPF goes bananas when changing parameters too abrubtly

Harvie, 2021-02-02:
Hello, i have troubles with LPF. It works well for a while, but when i use 

Here are relevant parts of code. This actualy works because i use monotonic cutoff value derived from `millis()`. But when i use `mozziAnalogRead(0)` as control signal for the cutoff, the filter can get out of whack and never stabilized again if there is too much noise. It works for a while as expected, but too abrubt changes in control signal switch the filter to some mode that it only outputs screechin noises remotely resembling the input signal. I understand that this might be caused by resonance being too high. However there is no way to make the filter behave again until i reboot the atmega.

Is this a bug or am i misusing something?

```C
LowPassFilter VCF;

void updateControl(){
  envelope.update();
  int mezi;
  mezi = ((millis()/4)%255); //DOES WORK NICELY
  //mezi = mozziAnalogRead(0)/4; //CAUSES ISSUES
  if(mezi < 10) mezi =10;
  VCF.setCutoffFreqAndResonance(mezi, 250);
}

AudioOutput_t updateAudio(){
  envval = envelope.next();
  return MonoOutput::from16Bit(envval * (VCF.next(aSin.next()/2)/2));
}
```

sensorium, 2021-02-02:
Hi Tomas,

try reducing the level of input to the filter even further... any distortion seems to through the filter into chaos.
I just tried a test with a rapidly changing cutoff freq which caused the filter to screech. then reduced the signal to the filter and it handled it OK.

You could possibly rewrite the fixed point maths in LowPassFilter.h to provide more headroom - at the cost of more processing time.

It might also be worth trying StateVariableFilter in case it's better behaved...

Harvie, 2021-02-03:
I am looking for that juicy ringing resonation sound, but even at full resonation all i can get is clicking/clipping anyways... So i guess there is no point of having it set so high...

I've lowered resonance from 250 to 160 and it seems to be more stable. (190 was still too much).

I think there should be some safeguard features added to the filter...

 * `LowPassFilter.reset()` method to manualy reset the filter to the initial state (this would be quite OK if i can reset the filter after each note, or each time  there is a silence). This seems to be an easy workaround to add.

 * Some "idle" handling code which will ensure that filter will gradualy ramp down and silence its internal state once the input is quiet (or zero) for a while.

Harvie, 2021-02-03:
BTW i am looking for 18dB filter.

State Variable filter is said to be 12db.
LowPassFilter docummentation does not specify the attenuation properties.

sensorium, 2021-02-03:
> BTW i am looking for 18dB filter. Not sure how to achieve that.

Sounds like you might like to write or adapt your own filter!  There are lots of dsp examples out there.  You can simply implement it in updateAudio(), though even better if you write a Mozzi class to share...

Harvie, 2021-02-03:
> Sounds like you might like to write or adapt your own filter! 

BTW there is online filter generator: http://t-filter.engineerjs.com/
Maybe we can make simple class that will be able to take data generated by that.
But i need voltage controlled resonant filter, so there's probably no use for that.

> though even better if you write a Mozzi class to share...

Yes i always prefer to go upstream so i don't have to track the development in my branch :-D

sensorium, 2021-02-07:
Here's something from https://beammyselfintothefuture.wordpress.com/2015/02/16/simple-c-code-for-resonant-lpf-hpf-filters-and-high-low-shelving-eqs/

Just a quick try tacking this at the end of Mozzi's LowPassFilter example gives some interesting results - you might need to play around to get what you want...

int lastOutput;
int momentum;
float cutoff = 6.35;
float reson = .005; // try v. low numbers here

AudioOutput_t updateAudio(){
    int distanceToGo = aCrunchySound.next() - lastOutput;
    momentum += ((int) ((float) distanceToGo*cutoff))<<2; // Lower / higher number here will lower / raise the cutoff frequency
    lastOutput += (int)((momentum + distanceToGo)*reson);//  Higher number here (max 1) lessens resonance
  return MonoOutput::from8Bit(lastOutput>>1); // >>1 when distorting
}