sensorium/Mozzi #117: ADSR: Allow unlimited sustain or add disable flag to ADSR

mark-orion, 2021-02-06:
Please add a feature that does allow either a sustain phase without any time limit, or a flag that allows disabling the envelope generator. This would allow easy transition from ADSR modulated notes to a continuous drone.

tomcombriat, 2021-02-10:
Hi,
Since PR #95 you can add a template parameter to the ADSR. Setting this parameter to `unsigned long` should let you having phases times for the ADSR reaching several days (from my basic calculation, around 163 days, even if the ADSR is updated at audio rate). Do you really need more than that?

mark-orion, 2021-02-12:
> Hi,
> Since PR #95 you can add a template parameter to the ADSR. Setting this parameter to `unsigned long` should let you having phases times for the ADSR reaching several days (from my basic calculation, around 163 days, even if the ADSR is updated at audio rate). Do you really need more than that?

Thanks for pointing this out. Changed the code and it works excellent. As I have no plans to perform for more 163 days without sleep this is indeed more than enough sustain time. I am closing this bug now.

closed 2021-02-12 by mark-orion