sensorium/Mozzi #119: Error compilation arduino

mrWang-t, 2021-02-21:
when replacing the code
`// # define AUDIO_MODE STANDARD
#define AUDIO_MODE STANDARD_PLUS
// # define AUDIO_MODE HIFI`
on
`// # define AUDIO_MODE STANDARD
// # define AUDIO_MODE STANDARD_PLUS
#define AUDIO_MODE HIFI`
the code stops compiling and many errors appear
tried on arduino pro and nano 

In file included from E:\!Main Folder\PROJECT\Arduino Lib\libraries\Mozzi-master/MozziGuts.h:210:0,
                 from E:\!Main Folder\PROJECT\Arduino\Mozzi\Audiophonic-Workbench-main\APW-Sketch\APW-Sketch.ino:24:
E:\!Main Folder\PROJECT\Arduino Lib\libraries\Mozzi-master/AudioOutput.h: In static member function 'static StereoOutput StereoOutput::fromNBit(uint8_t, int16_t, int16_t)':
E:\!Main Folder\PROJECT\Arduino Lib\libraries\Mozzi-master/AudioOutput.h:48:37: error: 'AUDIO_BITS' was not declared in this scope
 #define SCALE_AUDIO(x,bits) (bits > AUDIO_BITS ? (x) >> (bits - AUDIO_BITS) : (x) << (AUDIO_BITS - bits))
                                     ^
E:\!Main Folder\PROJECT\Arduino Lib\libraries\Mozzi-master/AudioOutput.h:100:97: note: in expansion of macro 'SCALE_AUDIO'
   static inline StereoOutput fromNBit(uint8_t bits, int16_t l, int16_t r) { return StereoOutput(SCALE_AUDIO(l, bits), SCALE_AUDIO(r, bits)); }
                                                                                                 ^~~~~~~~~~~
E:\!Main Folder\PROJECT\Arduino Lib\libraries\Mozzi-master/AudioOutput.h:48:37: note: suggested alternative: 'AUDIO_BIAS'
 #define SCALE_AUDIO(x,bits) (bits > AUDIO_BITS ? (x) >> (bits - AUDIO_BITS) : (x) << (AUDIO_BITS - bits))
                                     ^
E:\!Main Folder\PROJECT\Arduino Lib\libraries\Mozzi-master/AudioOutput.h:100:97: note: in expansion of macro 'SCALE_AUDIO'
   static inline StereoOutput fromNBit(uint8_t bits, int16_t l, int16_t r) { return StereoOutput(SCALE_AUDIO(l, bits), SCALE_AUDIO(r, bits)); }
                                                                                                 ^~~~~~~~~~~

sensorium, 2021-02-24:
Hi @tomcombriat and @tfry-git, I'm not familiar enough with the recent changes to audio output to quickly understand this issue - does something need to be changed in the avr code for hifi to work now?

tomcombriat, 2021-02-24:
Hi!
@tfry-git will know better, but I would be on a:
`#define AUDIO_BITS 14` missing in AudioConfigHiSpeed14bitPwm.h!

@mrWang-t Maybe you could try to add it, and if it does work we can correct it in the library!

mrWang-t, 2021-02-24:
thank!
many bugs fixed
however, the following remained

C:\Users\Roman\Documents\Arduino\libraries\Mozzi-master\MozziGuts.cpp: In function 'void __vector_7()':
C:\Users\Roman\Documents\Arduino\libraries\Mozzi-master\MozziGuts.cpp:676:3: error: 'defaultAudioOutput' was not declared in this scope
   defaultAudioOutput();
   ^~~~~~~~~~~~~~~~~~
C:\Users\Roman\Documents\Arduino\libraries\Mozzi-master\MozziGuts.cpp:676:3: note: suggested alternative: 'bufferAudioOutput'
   defaultAudioOutput();
   ^~~~~~~~~~~~~~~~~~
   bufferAudioOutput
exit status 1

tomcombriat, 2021-02-24:
Alright, that's a start!

@tfry-git If I'm correct, this line is missing in several AudioConfig files (`AudioConfigStandardPlus.h` and `AudioConfigStandard9bitPwm.h` (if still used)). Or I am missing something?

I could not find easily the root of this new error, probably a missing precompilator test around line 354 of MozziGuts.h, but as I do not have hardware right now I cannot test.

Will try to come back to this soon (if Thomas has not answered in between)!

mrWang-t, 2021-02-24:
now an error occurs which is not related to #define AUDIO_BITS 14
now there is some problem with the function call
defaultAudioOutput.
you can check it by pressing compilation, without having to download it into real hardware
Спасибо )))

tomcombriat, 2021-02-24:
> you can check it by pressing compilation, without having to download it into real hardware

You are right, however:
  - successful compilation does not necessary means correct behavior
  - I am currently on an hardware device (namely a phone) that does not support compilation.

In the meantime, as a wild guess, you could try replace line 354 of `MozziGuts.cpp` from
`#if (AUDIO_MODE == STANDARD) || (AUDIO_MODE == STANDARD_PLUS) || IS_STM32()`
to
`#if (AUDIO_MODE == STANDARD) || (AUDIO_MODE == STANDARD_PLUS) || IS_STM32() || (AUDIO_MODE == HIFI)`

If that does work, I (or someone else) will commit these changes to the library.

Best,

mrWang-t, 2021-02-24:
did not help. I will look forward to your chance to fix this.

And you can use https://github.com/techno-womble/Audiophonic-Workbench/blob/main/APW-Sketch.ino
to the full

All the best!

tfry-git, 2021-02-26:
Hi!

Sorry for the delay. I think this is now fixed, but I did not test on real hardware (a bit short on time, ATM).

Audio-modes STANDARD and STANDARD_PLUS do not need an AUDIO_BITS define, as these are special-cased (they're 8 bits and - err - a bit), although it may make sense to add that, for consistency.

tomcombriat, 2021-02-26:
> Audio-modes STANDARD and STANDARD_PLUS do not need an AUDIO_BITS define, as these are special-cased (they're 8 bits and - err - a bit), although it may make sense to add that, for consistency.

Oh ok! I only spotted the difference without trying to compile, I missed something somewhere… I never used HIFI mode, that's probably why this went through the test phase. Thanks for helping me to get more into the core of Mozzi's code!

mrWang-t, 2021-02-26:
many thanks! everything works correctly!

closed 2021-02-27 by tfry-git