sensorium/Mozzi #121: Won't Compile

pettefar, 2021-04-21:
In file included from /home/nick/sketchbook/libraries/Mozzi/Oscil.h:24:0,
                 from APW.ino:25:
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:22:5: warning: identifier ‘static_assert’ is a keyword in C++11 [-Wc++0x-compat]
     static_assert(sizeof(T) == 1 || sizeof(T) == 2 || sizeof(T) == 4 || sizeof(T) == 8, "Data type not supported");
     ^
APW.ino:28:0: warning: "AUDIO_RATE" redefined
In file included from /home/nick/sketchbook/libraries/Mozzi/MozziGuts.h:157:0,
                 from APW.ino:24:
/home/nick/sketchbook/libraries/Mozzi/mozzi_config.h:62:0: note: this is the location of the previous definition
 #define AUDIO_RATE AUDIO_RATE_PLATFORM_DEFAULT
 ^
APW.ino:29:0: warning: "CONTROL_RATE" redefined
In file included from APW.ino:24:0:
/home/nick/sketchbook/libraries/Mozzi/MozziGuts.h:46:0: note: this is the location of the previous definition
 #define CONTROL_RATE 64
 ^
In file included from /home/nick/sketchbook/libraries/Mozzi/MozziGuts.h:210:0,
                 from APW.ino:24:
/home/nick/sketchbook/libraries/Mozzi/AudioOutput.h: In function ‘uint32_t pdmCode8(uint16_t)’:
/home/nick/sketchbook/libraries/Mozzi/AudioOutput.h:187:37: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
   static const byte fast_pdm_table[]{0,  0b00010000, 0b01000100,
                                     ^
In file included from /home/nick/sketchbook/libraries/Mozzi/Oscil.h:24:0,
                 from APW.ino:25:
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h: In function ‘T mozzi_pgm_read_wrapper(const T*)’:
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:22:114: error: there are no arguments to ‘static_assert’ that depend on a template parameter, so a declaration of ‘static_assert’ must be available [-fpermissive]
     static_assert(sizeof(T) == 1 || sizeof(T) == 2 || sizeof(T) == 4 || sizeof(T) == 8, "Data type not supported");
                                                                                                                  ^
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:22:114: note: (if you use ‘-fpermissive’, G++ will accept your code, but allowing the use of an undeclared name is deprecated)
In file included from /home/nick/sketchbook/libraries/Mozzi/Oscil.h:24:0,
                 from APW.ino:25:
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h: In function ‘T mozzi_pgm_read_wrapper(const T*) [with T = double]’:
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:35:162: error: ‘static_assert’ was not declared in this scope
     static_assert(sizeof(uint64_t) == sizeof(double) || sizeof(float) == sizeof(double), "Reading double from pgmspace memory not supported on this architecture");
                                                                                                                                                                  ^
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:38:17: warning: extended initializer lists only available with -std=c++11 or -std=gnu++11
         return u{mozzi_pgm_read_wrapper((uint64_t*) address)}.d;
                 ^
In file included from /home/nick/sketchbook/libraries/Mozzi/Oscil.h:24:0,
                 from APW.ino:25:
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h: In instantiation of ‘T mozzi_pgm_read_wrapper(const T*) [with T = long long unsigned int]’:
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:38:60:   required from here
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:22:18: error: ‘static_assert’ was not declared in this scope
     static_assert(sizeof(T) == 1 || sizeof(T) == 2 || sizeof(T) == 4 || sizeof(T) == 8, "Data type not supported");
                  ^
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h: In instantiation of ‘T mozzi_pgm_read_wrapper(const T*) [with T = signed char]’:
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:48:38:   required from ‘T FLASH_OR_RAM_READ(T*) [with T = const signed char]’
/home/nick/sketchbook/libraries/Mozzi/Oscil.h:308:41:   required from ‘int8_t Oscil<NUM_TABLE_CELLS, UPDATE_RATE>::readTable() [with unsigned int NUM_TABLE_CELLS = 2048u; unsigned int UPDATE_RATE = 32768u; int8_t = signed char]’
/home/nick/sketchbook/libraries/Mozzi/Oscil.h:93:19:   required from ‘int8_t Oscil<NUM_TABLE_CELLS, UPDATE_RATE>::next() [with unsigned int NUM_TABLE_CELLS = 2048u; unsigned int UPDATE_RATE = 32768u; int8_t = signed char]’
APW.ino:100:27:   required from here
/home/nick/sketchbook/libraries/Mozzi/mozzi_pgmspace.h:22:18: error: ‘static_assert’ was not declared in this scope
In file included from APW.ino:24:0:
/home/nick/sketchbook/libraries/Mozzi/MozziGuts.h: At global scope:
/home/nick/sketchbook/libraries/Mozzi/MozziGuts.h:377:13: warning: ‘void setupTimer2()’ declared ‘static’ but never defined [-Wunused-function]
 static void setupTimer2();
             ^

tfry-git, 2021-04-23:
Are you using a really old version of Arduino / GCC by any chance?

pettefar, 2021-04-23:
It’s the latest on the Raspberry Pi.

MfG,

Nick

tfry-git, 2021-04-25:
Is that 1.0.5, then? That would be really old, indeed. However, the nets seem to agree that you can use the more recent "upstream" versions of arduino provided, here: https://www.arduino.cc/en/software (compiled for Linux/ARM).

At any rate those errors and warnings point to the compiler in use not understanding or not using c++11.

pettefar, 2021-04-25:
Das ist perfekt!  Vielen dank Thomas!

Gruesse aus Irland!

Nick

/home/nick/sketchbook/APW/APW.ino:28:0: warning: "AUDIO_RATE" redefined
 #define AUDIO_RATE 32768

In file included from
/home/nick/Arduino/libraries/Mozzi-master/MozziGuts.h:157:0,
                 from /home/nick/sketchbook/APW/APW.ino:24:
/home/nick/Arduino/libraries/Mozzi-master/mozzi_config.h:62:0: note: this
is the location of the previous definition
 #define AUDIO_RATE AUDIO_RATE_PLATFORM_DEFAULT

/home/nick/sketchbook/APW/APW.ino:29:0: warning: "CONTROL_RATE" redefined
 #define CONTROL_RATE 128

In file included from /home/nick/sketchbook/APW/APW.ino:24:0:
/home/nick/Arduino/libraries/Mozzi-master/MozziGuts.h:46:0: note: this is
the location of the previous definition
 #define CONTROL_RATE 64

Sketch uses 6766 bytes (22%) of program storage space. Maximum is 30720
bytes.
Global variables use 837 bytes (40%) of dynamic memory, leaving 1211 bytes
for local variables. Maximum is 2048 bytes.

closed 2021-04-26 by tfry-git