sensorium/Mozzi #123: Cleanup MozziGuts.cpp

pschatzmann, 2021-05-08:
I was wondering if there are any plans to cleanup the situation in MozziGuts.cpp.  I frankly think in the current state it is unmaintainable and maybe a better way forward would be to split things up into separate platform specific implementation files. I was considering to add support for the Raspberry Pico, but in the current state this would make the situation only worse.

I would also prefer to have the related functions in proper classes. This would allow to come up easily with some alternative implementations which might even apply cross platform. Something like: 
Mozzi mozzi;
mozzi.start();
and mozzi.audioHook()...

I would be willing to spend my time working on this.

tfry-git, 2021-05-09:
Hi!

You are right, MozziGuts.cpp is pretty terrible - and it used to be worse. Somewhat recently, we modularized and split out a lot of code into AudioOutput.h, which really helped (compare the present audioHook() - https://github.com/sensorium/Mozzi/blob/89317c3cdd51f8ae26c49f605290cefe416ad51a/MozziGuts.cpp#L325 - with the earlier version - https://github.com/sensorium/Mozzi/blob/3d320a0c53d06201221e5d86e606775b8ffcb394/MozziGuts.cpp#L383). We still keep all platforms inside one code file, in AudioOutput.h, but arguably, that is not the most important detail, compared to having a manageable and documented breakup in the first place.

It is more than true that there is more potential for this sort of work in MozziGuts.cpp. Importantly, code for reading audio input and code for setting up the required platform specific timers is still rather hard to follow.

I sort of like your idea of organizing all this inside a class, _but_ we need to be very, very much on the lookout, not to incur additional code cycles on the AVR 8-bitters. As long as we can be sure that all such syntactic sugar will be optimized away by the compiler, I'm all for it. However, true virtual functions will probably be out of reach, for one thing.

Long story short: Work in that direction will be very welcome, and definitely possible, but you will have to be careful in your selection of tools.

pschatzmann, 2021-05-09:
Thanks for your feedback. 

As an excercise I kind of tried to do the split out for the ESP32 and finally came to the following conclusions:
- It is almost impossible to do this w/o introducing any errors - so I need to think about building some tools which would automate this
- Even with the half finished version it is much easier to reason about the code and find potential problems: e.g. 
i2s_driver_uninstall() is missing in stopMozzi().
- I think we should go for an incremental approach where we can switch the different Processor implementations one by one.

pschatzmann, 2021-05-10:
Hallo, I used unifdef to split the MozziGuts.cpp into different implementation files.
MozzGuts will be Replaced by Mozzi.h, Mozzi.cpp, MozziEsp32.cpp etc

Like for Serial I created an global Mozzi object. So the new syntax will be Mozzi.start(), Mozzi.stop().  I am still supporting the old commands - though. 

Maybe you find the time to provide some feed back.

https://github.com/pschatzmann/Mozzi

tfry-git, 2021-05-12:
I only had a cursory look so far. First thoughts:

- I'm inclined to think that your method may actually be the way to go. Yet, I'm not quite happy the duplicated chunks of code. E.g. audioHook() is (almost) platform independent, but now it is split into eight different versions, which is creating a new orthogonal problem. This will need some love to sort out.
- For the MozziClass: I like the looks of that, too. But without having tested, I believe your current setup will induce quite some overhead (which will not even be measurable on RPi, but may easily prove prohibitive on AVR8). However, I suppose this will always remain a singleton, and also we don't really have to support different implementations of it _in a single sketch_. Thus maybe something looking almost as nice and clean can be had by using all static member functions?

pschatzmann, 2021-05-12:
Thanks for your feed back

- I agree with your statement for the audioHook - It is very similar across the environments. I am currently thinking about solving the stereo hack by adding multi channel support and this is exactly impacting the audioHook across all environments and might have an impact on performance.
- It is my expectation that the MozziClass or any alternative Output implementations (e.g. MozziOut which provides data as Arduino Stream) should be valid only once in one sketch and I am not worried about lost performance in functions which are called only once. I tried hard that I did not add any overhead in anything that is executed in the loop() or in a timer. But I might have overlooked something...

pschatzmann, 2021-05-12:
Here would be an example:

https://github.com/pschatzmann/Mozzi/blob/master/examples/13.External_Audio_Output/Serial_Output/Serial_Output.ino

tfry-git, 2021-05-13:
Re performance: My worry is specifically about audioHook(), getAudioInput(), mozziMicros(), and audioTicks(). Turning these into regular class members will mean that, internally, a ```this``` pointer has to be passed in each function call. That'll be four additional cycles per call on AVR, which is starting to be something when trying to squeeze the best possible AUDIO_RATE out of a 16Mhz CPU.

Yet, again, I think that problem can easily be remedied by declaring all these functions static. Usage would remain the same, but the ```this``` pointer falls away, again. (As a second measure, it will probably be necessary to inline the legacy support functions - ugly as that may be - in order to avoid actually introducing additional calls).

----

Re the #ifdef hell: IMO, this is really the first thing to address, and at any rate, I'd like to move forward step-by-step, instead of having to think about all these changes at once. Do you think, you could create a separate PR for just the split-up of MozziGuts.cpp alone, and then we can first focus on sorting out all details WRT to that?

pschatzmann, 2021-05-14:
No worries... I am struggling a little bit with the logic round BYPASS_MOZZI_OUTPUT_BUFFER:
Can you confirm that this has just been introduced for the ESP32 and ESP8266 to manage the output to I2S and is not relevant for any other environment - My feeling is that we can get rid of this totally by having the split up....

tfry-git, 2021-05-14:
BYPASS_MOZZI_OUTPUT_BUFFER is currently used just for ESP32 and ESP8266 I2S output. Correct. Yet, it might also be interesting for STM32 and possibly others, though. As far as I know these can also be coerced to emulate I2S with a hardware DMA transfer buffer, which would be a useful extension.

Beyond that, it may be used in combination with EXTERNAL_AUDIO_OUTPUT, though I would be somewhat surprised to learn of any project already doing so.

pschatzmann, 2021-05-14:
Thanks for your feed back!

I think we should try to separate different output scenarios into separate classes as well. I see the primary goal of MozziClass to output to pins !

Any other scenario should not be shoehorned into this.

By the way I think it would have been a better decision to use the buffer for the ESP32 I2S output as well, and I am surprised that you did not have any performance issues with the current approach ... but I don't want to open this can of worm...

tfry-git, 2021-05-14:
> By the way I think it would have been a better decision to use the buffer for the ESP32 I2S output as well, and I am surprised that you did not have any performance issues with the current approach ... but I don't want to open this can of worm...

Well it is using a buffer, but in hardware/DMA (and in fact, that is the same size as the default mozzi buffer), so there is no point in adding a software buffer on top of that.

tfry-git, 2022-03-01:
I essence this was addressed in https://github.com/sensorium/Mozzi/pull/145 . Not saying there isn't room for further improvement, but closing this ticket.

closed 2022-03-01 by tfry-git