sensorium/Mozzi #130: Problems with compiler optimization options / Mozzi example sketches not working

ChrisVeigl, 2021-06-26:
hi,

when trying Mozzi with an Arduino UNO (ATmega328), I found that many examples did not work as intended,
even _01.Basics->Control_Gain_ didn't actually change the gain of the sinewave!

After adding a _Serial.print (gain)_ in the audio loop it worked,
which indicated that the gcc/g++ compiler incorrectly strips away the variable access during optimization (without the print call).

Strange enough, a _volatile_ declaration of _gain_ did not help, but when i changed the overall optimization options in the platform.txt file to O1, all example files worked correctly!

did somebody experience that problem ?
(i am using Arduino IDE 1.8.13 - current version 1.8.15 does not seem to be different in this regard).

here the line in platform.txt which i had to modify (_Os_ was changed to _O1_):

`compiler.cpp.flags=-c -g -O1 {compiler.warning_flags} -std=gnu++11 -fpermissive -fno-exceptions -ffunction-sections -fdata-sections -fno-threadsafe-statics -Wno-error=narrowing -MMD -flto`

(note that O2 and O3 did not work either, and O0 produced huge code size which prevented larger sketches from working)

sensorium, 2021-07-04:
Hi Chris,
thanks for the report.  An extra ")" had found its way into the Control_Gain example, which stopped it compiling here.  That's now fixed in the repo, and the sketch works for me.
I tried a few examples, but didn't come across any others that were broken.  Do let me know which ones aren't working, if you notice more problems.
I'll try to find time to look at the optimisation issues.

ChrisVeigl, 2021-07-04:
thank you for the reply - and for fixing the example code :-)

there must be some other reason though, because on my end i could compile everything without problems, but many sketches did not work as expected (unless i change the compiler options to -O1, which unfortunalty can only be accomplished in the Arduino "IDE" by changing the platform.txt file, thereby influencing also the optimization settings for other sketches ...)

I tried on different machines, same problem. 
But I did not test different Arduino IDE versions... so maybe related to that... 

Which version do you use ?

sensorium, 2021-07-04:
version 1.8.13 at the moment

ChrisVeigl, 2021-07-05:
I am using the same version...
strange enough  !

attached you find the hex file compiled with -Os (standard, not working for me in various sketches / scenarios) and a version compiled with -O1 (works for me).

[hexfiles.zip](https://github.com/sensorium/Mozzi/files/6763868/hexfiles.zip)

if this is not relevant for other people
 we can close the issue ...

tomcombriat, 2023-01-31:
Should we close this?

ChrisVeigl, 2023-01-31:
i still have the issue with Arduino IDE 1.8.13;
i did not try with IDE version 2.x by now.
feel free to close the issue if others do not have it!