sensorium/Mozzi #131: 'MonoOutput' has not been declared

emirgo, 2021-07-23:
Hi, 

I downloaded the latest release (not pre-release) and currently running a firmware that has MonoOutput class to it. However, compilation fails due to error _'MonoOutput' has not been declared_

Any help would be appreciated.

sensorium, 2021-07-25:
Hi Emir,

MonoOutput is part of the current code on Github, it's not used in the release.  It's been a long time since a release, and I recommend using the developing code on Github.

Tim

kanduvisla, 2021-12-09:
Hi, I also get this error when trying to compile for Arduino Nano Every.

I downloaded the ZIP file from GitHub (not from releases), and I checked and it looks like the `master`-branch has the latest code on it. But still I'm getting the error:

```
In file included from /.../Arduino/libraries/Mozzi-master/MozziGuts.h:207:0,
                 from /.../Arduino/mozzi_tst/mozzi_tst.ino:1:
/.../Arduino/libraries/Mozzi-master/AudioOutput.h: In function 'void audioOutput(MonoOutput)':
/.../Arduino/libraries/Mozzi-master/AudioConfigStandardPlus.h:20:41: error: 'OCR1A' was not declared in this scope
 #define AUDIO_CHANNEL_1_OUTPUT_REGISTER OCR1A
                                         ^
/.../Arduino/libraries/Mozzi-master/AudioOutput.h:365:3: note: in expansion of macro 'AUDIO_CHANNEL_1_OUTPUT_REGISTER'
   AUDIO_CHANNEL_1_OUTPUT_REGISTER = f.l()+AUDIO_BIAS;
   ^~~~~~~~~~~~~~~~~~~~~~~~~~~~~~~
exit status 1
Error compiling for board Arduino Nano Every.
```

If I try to compile for a "regular" Arduino Nano it shows no errors.