sensorium/Mozzi #142: ESP32 Input audio

ichin23, 2021-12-20:
I'm very excited with this lib because I want to do a guitar with multi effects. But I just have an ESP32, and I think don't have audio read because the horrible integrated ADC. Is possible read audio signal with ADS1115 and use this library to get effects? How can I do without the getAudioInput()?

pschatzmann, 2023-02-28:
I was playing around with ADS1015. These two boards are very similar, differing only in resolution and speed.  The ADS1115 has higher resolution and the ADS1015 has a higher sample rate.

However if the ADS1015 is much too slow for audio you don't have a chance with the ADS1115!
see https://www.pschatzmann.ch/home/2021/11/12/the-big-ads1015-audio-failiure/

If you use a proper ADC or the built in ADC via I2S you would still need to extend this library a bit because the getADCReading() method is not implemented in [MozziGuts_impl_ESP32.hpp](https://github.com/sensorium/Mozzi/blob/master/MozziGuts_impl_ESP32.hpp)

pschatzmann, 2023-02-28:
I was conidering to add the following implementations for the getADCReading() method:

- get data from the internal DAC via I2S
- get data from an external DAC via I2S

At first sight this seems to be something very easy to do, so I was wondering if I miss anything ? After all there must be a reason why this does not exist yet...

tomcombriat, 2023-02-28:
Hi @pschatzmann , 
Feel free to try! I personally do not have a ESP32 to play around to try to implement that, so I do not know if there is a caveat or not ;)

tfry-git, 2023-02-28:
I think analog input, and audio rate analog input in particular has simply been a bit of a step child, so far...

I suppose the main difficulty will be to wrap your mind around the convoluted analog read logic. Essentially all this is designed around the need to allow for a non-blocking round-robin sampling on the Arduino Uno. That may prove to be a bit of a burden, when trying to implement something new, at least when trying to support input from more than one channel. I wouldn't mind a redesign of the internals, here, if you happen to have a clever plan for that. The only functions intended for end use are mozziAnalogRead(), and getAudioInput(), as far as I am aware.

pschatzmann, 2023-02-28:
mozziAnalogRead() is currently not on my radar and I mainly target the getAudioInput()
My gut feeling is that the big work has already been done by Espressif and most of the time the call will be non blocking because there is data in the I2S buffer. I have a first DRAFT implementation ready which the following option: 
```
// Audio IO options
#define INTERNAL_DAC 1 // output using internal DAC via I2S, output on pin 26
#define PT8211_DAC 2   // OBSOLETE: use I2S_DAC instead
#define PDM_VIA_I2S 3  // output PDM coded sample on the I2S data pin (pin 33, by default, configurable, below)
#define I2S_DAC 2      // output using an external DAC via I2S
#define I2S_DAC_AND_INTERNAL_ADC 4  // output using an external DAC via I2S and input via internal ADC
#define I2S_DAC_AND_I2S_ADC 5  // output using an external DAC , input with exteran ADC - both via I2S
#define INTERNAL_DAC_AND_I2S_ADC 6 // output using internal DAC via I2S, output on pin 26, input via external ADC
```
One final question ? Does getAudioInput(); need to return values in the range of 0-1023 of am I free to return the same data format that is used on the output side ? 
The scenario that I have in mind is a I2S based Guitar Effects pedal that gets the data as signed 16 bits and provides the output with signed 16 bits