sensorium/Mozzi #143: Problem installing under Arduino ise 2.0.0-rc

dsssssssss9, 2022-01-09:
I am trying to install a .zip library on IDE 2.0.0-RC2 ( portable version )

i get an error message...

Error: Request installZip failed with message: 13 INTERNAL: Library install failed: moving extracted archive to destination dir: getting stat info for /tmp/arduino-cli-lib-484157649/Mozzi-master/Mozzi: stat /tmp/arduino-cli-lib-484157649/Mozzi-master/Mozzi: too many levels of symbolic links

The folder is & contents are in my libraries folder but the IDE does not list it as installed

Installs fine under IDE 1.8.15 

Any help gratefully received

coryalder, 2022-09-14:
I am having the same issue, arduino ide 2.0.0, download the zip file of the master branch code, select Add Zip Library... in Arduino, result is the same error.

karlsoderby, 2022-09-22:
@coryalder for some reason, it worked by manually unzipping and dragging the "Mozzi" folder directly into your libraries folder. I had the same problem.

storrdev, 2022-11-14:
I believe this PR has fixed this issue: https://github.com/sensorium/Mozzi/pull/156

tomcombriat, 2022-11-15:
Closing this as solved by #156