sensorium/Mozzi #153: Wavefolder

cwmoriarty, 2022-06-16:
Has anyone considered developing a wavefolder for Mozzi?

See https://noiseengineering.us/blogs/loquelic-literitas-the-blog/getting-started-wavefolders

If there is interest, I could try my hand at creating one.

tomcombriat, 2022-06-16:
Hi,
I personally tried some time ago, but stopped when I wanted it not to be "spiky" upon folding, a bit more like an analog behavior…
You are of course welcome to try and submit something through a PR if you get some nice results!

cwmoriarty, 2022-06-16:
"spiky" with the sharp turn? Maybe a bit of slew would help?

If you have any of the old code, please send it along.

sensorium, 2022-06-17:
One (quick?) approach is to try Mozzi's WaveShaper class which uses a table to determine how the incoming wave gets shaped (or folded).  The demo is in the synthesis examples.  Scripts for producing tables are in Mozzi/extras/python (see table_generator_template.py), or you could draw tables in Audacity, save as raw 8bit and convert with char2Mozzi.py.

tomcombriat, 2022-06-17:
@cwmoriarty Yes, or just a LowPassFilter after the wavefolding. I have not trace of what I did, that was really a quick test. I think the part that can be tricky is to find, in an efficient way, how many times it has to be folded (if the folder is harsh enough so that it bounces back a few times). I am happy to help if you get started!

@sensorium Yes, that's an idea but if I understand correctly, using the WaveShaper prevents the levels at which the folding will occur to be changed at run-time, no? A big plus comes for waveshapper when levels can be bound to an envelope or so…

sensorium, 2022-06-17:
Yes, you're right Tom, using a single fixed table won't enable run-time changes in the mapping.
Just in case it's any use at all, I came across these quick sketches of wave folding and shaping in Pd (though it sounds like you're way ahead already):
folding: https://youtu.be/f2s3mAqLjkg
shaping: https://youtu.be/sg2TpxQ9CbU

closed 2023-01-12 by tomcombriat