sensorium/Mozzi #155: Teensy 4.1 high definition circuitry

PierreBanwarth, 2022-09-13:
Seen in the AudioConfigTeensy4.h 
`
#define AUDIO_CHANNEL_1_PIN A8
#define AUDIO_CHANNEL_2_PIN A9 
`
are A8 and A9 are for stereo sound ?

I'm looking for a way to make High definition audio sound.

Is there an arduino like option to use the high defintion circuitry ?

https://sensorium.github.io/Mozzi/learn/output/ 

Sorry if it's not the good place

tomcombriat, 2022-09-13:
Hi,

There is no HiFi mode available for Teensy. This probably originates from the original port for Teensy 3 as this platform already had a 12bits DAC embedded, but this is not the case for Teensy 4.

A good way to have good audio output is to use an external DAC (like the PT8211 for instance). Alternatively, I do not see why you could not send the MSB to channel1 and the LSB to channel2 in stereo mode and then recombine them with a similar HiFi circuit than for the Arduino. Never tried though.

Hope this helps,

tomcombriat, 2025-01-14:
Closing this as this is fairly old and seems to be stalled. Please feel free to reopen if that was not the case!

closed 2025-01-14 by tomcombriat